supermerill/SuperSlicer #2196: Gcode preview shows wrong first layer height (and width)

JohanAR, 2022-01-08:
### What happened?

Opened attached .gcode file in the preview. For some reason SuSl thinks the first layer is 0.15mm thick, which also causes it to calculate line widths wrong.

[image]

[image]

There is a "G0 Z0.15" in the custom start code, but then the gcode clearly moves to Z0.3 before it begins the main print.

### Project file & How to reproduce

[AI3MSPRO_test_matrix_v1.gcode.zip](https://github.com/supermerill/SuperSlicer/files/7833314/AI3MSPRO_test_matrix_v1.gcode.zip)

### Version

2.3.57.9

### Operating system

Xubuntu 21.10

supermerill, 2022-01-08:
I think it's because 0.3-0.15 = 0.15

I'll look to see if it's possible to fix it easily

JohanAR, 2022-01-09:
You mean it thinks the real first layer is being printed on top of the non-existing layer at 0.15mm?

Sounds like it would be difficult to make a perfect solution. E.g. someone could print a square at 0.2mm, and then at 0.4mm print both a second layer on top of the square, and a new square which would have a height of 0.4mm. But I suppose that's not something slicers do..

Are you parsing and using the ;LAYER:0 comments? Perhaps it would be possible to ignore anything before it?

Cura gcode preview behaves in the same way btw