supermerill/SuperSlicer #2197: [FR] "has_multiple_extruders" variable

DSPuzzles, 2022-01-08:
For people running IDEX or other multiple tool systems, it would be useful to have a `has_multiple_extruders` placeholder variable to feed into the print start g-code.

Current behaviour for printers with multiple extruders is as follows:
- Print uses one tool: no tool start g-code (ex: `T[initial_extruder]`) is added to the sliced g-code after the print start g-code.
- Print uses multiple tools: tool start g-code is issued for the first active extruder after the print start g-code.

This means that if the tool start g-code (which runs on tool changes) includes routines such as purging, they won't run at the start of single tool prints. The alternative is to include purging on the print start g-code, but then that would mean purging twice on multiple tool prints, since the tool start g-code would run as well.

A `has_multiple_extruders` variable would be a nice solution, since it would allow for deciding whether or not to run these routines (like purging) on the print start g-code depending on if the print uses one or more tools.

An alternative would be to always include a tool start g-code, even when the print has only one tool. This may lead to issues on some setups, but a "Run tool start g-code on single tool prints" checkbox could be added in the printer settings to give a way to turn this off. I'm not sure what the best way to handle single tool prints with tools set as "default" with no tool selected.

supermerill, 2022-01-31:
I'm more for `number_extruder_used` and `extruders_idx` (as an array of extruder id, the size of number_extruder_used)

DSPuzzles, 2022-01-31:
Yeah, I agree, storing the number of extruders is significantly more useful than storing whether it uses more than one, and can be easily used in macros.

However, I have since thought a bit more about the "Run tool start g-code on single tool prints" checkbox, and it seems like a good feature as well. It doesn't add in any extra problems, and (I think) it fits well with the rest of the settings. The only grey area is what to do with the default extruder, like I mentioned in the first post.

BrewNinja, 2024-02-29:
Prusa Slicer has is_extruder_used [](https://help.prusa3d.com/article/list-of-placeholders_205643). It would be super awesome to see either the original idea or the Prusa way integrated. Being able to preheat multiple extruders before you print is a must have. Thanks!

diode-ee, 2024-04-24:
Bump for the request of this feature.

jdmontgomer, 2024-04-27:
Have 6 tool heads on a voron 2.4 would love ss back

dweenz, 2024-08-10:
Would be great to use superslicer for my stealthchanger

Dumplap, 2024-08-10:
stealthchanger

SarraceniaWilds, 2024-09-30:
bump for request, I can't use my sv04 start gcode with superslicer and don't see any way to translate it with the available variables, mine uses is_extruder_used