supermerill/SuperSlicer #2203: print_host variables appears empty in GCode and post processing script contexts

wlhlm, 2022-01-09:
### What happened?

I'm trying to pass the printer host URL (`print_host` variable in superslicer) to a post processing script (via the `SLIC3R_PRINT_HOST` environment variable), but the variable appears to be empty. Doing some debugging, I've also tried to use the variable in a custom GCode context (start GCode) and it also appears empty there.

I've made sure to select a physical printer profile. The physical printer profile uses the Octoprint host type.

### Project file & How to reproduce

Here is my complete slicer config bundle: [wlhlm-print-host-bug-bundle.ini.gz](https://github.com/supermerill/SuperSlicer/files/7835351/wlhlm-print-host-bug-bundle.ini.gz)

Looking at the config bundle, I can see that each printer profile has an empty `print_host` variable. My guess is that there is an issue with superslicer not overwriting them with the corresponding `print_host` variable from the physical printer section (in case such a profile is selected when slicing).

### Version

2.3.58.0

### Operating system

Arch Linux

### Printer model

V2.4 (Klipper+moonraker)

supermerill, 2022-03-16:
These have been removed by prusa with the introduction of phsical printer.
have to find a way to re-add them, but only where it's allowed.

wlhlm, 2022-04-06:
Thank you for looking into fixing this. Unfortunately, this doesn't seem to quite work yet. `SLIC3R_PRINT_HOST` is still empty when the post-processor is called. I'm testing using the following script:

pp-env.sh
```
#!/usr/bin/env sh

env > /tmp/ss-pp.env
```

This is on SS built from commit https://github.com/supermerill/SuperSlicer/commit/810009e7f7844a72251c37694b3f8cc4ca7a562d

```
$ grep SLIC3R_PRINT_HOST /tmp/ss-pp.env
SLIC3R_PRINT_HOST=
```

supermerill, 2022-04-07:
Where did you find the doc about the SLIC3R_PRINT_HOST env variable?

wlhlm, 2022-04-07:
What I'm trying to do is to use the klipper_estimator utility (See #2226) as a post processor. This utility requires the URL of the printer to which the GCode is supposed to be uploaded. I was looking for a way to pass that URL without having to hardcode it.

Currently I have to use the following for `post_process`:
```
/usr/bin/klipper-estimator --config_moonraker_url http://192.0.2.100 post-process
```
I want SS to automatically replace the hardcoded `http://192.0.2.100` with the address from the currently selected physical printer. Reading the tooltip for `post_process`, it says that all the slicer settings are passed to the post processing script via environment variables. I then used the dummy post processing script I [posted above](https://github.com/supermerill/SuperSlicer/issues/2203#issuecomment-1090415268) to figure out which environment variables are set by SS. `SLIC3R_PRINT_HOST` seems like exactly the variable I want. Unfortunately it's empty as I said above.

If `print_host` (and thus `SLIC3R_PRINT_HOST`) are correctly initialized, I could wrap the `klipper-estimator` call above with the following script:
```
#!/usr/bin/env sh

/usr/bin/klipper-estimator --config_moonraker_url "${SLIC3R_PRINT_HOST}" post-process "$@"
```

closed 2022-07-14 by supermerill

wlhlm, 2023-01-18:
This is still not fixed in 2.4.58.5.