supermerill/SuperSlicer #2209: Gap between infill and perimiter when setting "Better bonding" > 0%

currazo, 2022-01-11:
### What happened?

When using the "Better bonding" option found under "Perimeters and shell", the sliced gcode leaves a gap between the infill and the perimiters. Clearly visible in both slicer and on actual print.

[image]

I guess that problem is that the infill is printed the same as when Better bonding = 0%.

### Project file & How to reproduce

Slice any model with Better bonding above 0% and it's clear that there's a gap between the infill and the perimters.

[BetterBonding.zip](https://github.com/supermerill/SuperSlicer/files/7849218/BetterBonding.zip)

### Version

2.53.57

Pavulon87, 2022-01-12:
#1065
> yes, the infill isn't moved. You can increase the infill/perimeter overlap if it's too far away.

currazo, 2022-01-12:
Aha, didn't see the option for infill/perimeter overlap but that should fix it. I guess it works as intended then based on the comments in the other issue.

closed 2022-01-12 by currazo

supermerill, 2022-01-12:
I think i did it with the idea of using "external perimeter first".