supermerill/SuperSlicer #2210: Can't get [solid infill] or [bridge infill] to cover entire surface area as if it were [top infill]

LeavOA, 2022-01-11:
### What happened?

Slicing a file with a large area, I would like to be able to control the bridge layers and the solid layers so that they cover 100% of the parts surface between perimeters, (in my mind, I imagine the slicer treating that layer similarly to a top layer, ignoring the fact that stuff is happening above.

I tried using height modifiers, to no avail.

Hopefully the attached gif works and explains the issue, if the attachment doesn't work, [see link here](https://i.imgur.com/Lv0dLUz.gif)

### Project file & How to reproduce

[PooTPHolder.zip](https://github.com/supermerill/SuperSlicer/files/7849796/PooTPHolder.zip)
[image]

### Version

Version 2.3.57

### Operating system

Windows 8.1

### Printer model

Delta with Duet2-wifi

LeavOA, 2022-01-12:
Found this issue: #1981 

Which led me to the "Anchor solid infill by Xmm" (under infill settings).

By setting it to a ridiculous amount, I was able to achieve the desired result (see below).

I still think it is worth having as an explicit option in the layer modifiers (e.g. "Infill entire layer") in order to create this behavior in a more straightforward way.

Maybe this exists? Or is this a feature request?

[image]
[image]

neophyl, 2022-01-12:
Another way to fix it would have been to turn OFF single perimeter on top layers.  Your original project attached with that single setting off slices in what I would consider the correct way in those areas.  I don't know why that setting causes so much unpredictability but if I ever see anything weird in a slice its always the first thing I try turning off.  Often it makes the issue go away.

LeavOA, 2022-01-12:
@neophyl Can you share what you got? cause the closest I can get is with the "solid infill every 1 layer" modifier, and it still creates disconnected infills.

[image]

[PooTPHolder_2.zip](https://github.com/supermerill/SuperSlicer/files/7852460/PooTPHolder_2.zip)

neophyl, 2022-01-12:
Maybe I should I worded the setting description better.  I changed a single setting, located in Print Settings>Quality>Only One perimeter>On Top Surfaces.  Turning it off results in the infill going under the perimeters.  However visually the print looks better with a single perimeter on those top layers.  As such you can apply a modifier on them to get it back without it messing up the other layers.  
I have attached a zip file with 2 projects included (both done with 2.3.57.9).  The first with the setting off and the other with the modifiers.
[PooTPHolder_neo.zip](https://github.com/supermerill/SuperSlicer/files/7852840/PooTPHolder_neo.zip)

LeavOA, 2022-01-14:
Thanks @neophyl, but that doesn't exactly capture what I'd like to do, since the infill doesn't _quite_ reach the perimeter (and the infill inside remains disconnected.

I think what I'm after could be very useful, but should be described in a separate FeatureRequest.

neophyl, 2022-01-14:
If you want to make it go completely over the layer then you have to set the growth of the infill. The setting you were doing before. Instead of percentage just set it as mm. You will need around 10-15mm growth to fill in the circular area.

closed 2022-01-16 by LeavOA