supermerill/SuperSlicer #2212: Perimeters and bridging under top solid infill breaks it

OverWorldD, 2022-01-12:
[image]
I have perimeters and a bridge under this top infill layer. Is there a way to make it fill the layer more uniformly? Bu some to me not understood reason it wants to fill the area over the bridge separately and that leaves an ugly gap.

neophyl, 2022-01-13:
Could you post your zipped up saved project file (.3mf from Susi) please.

OverWorldD, 2022-01-18:
Sure, here it is
[Line Select.zip](https://github.com/supermerill/SuperSlicer/files/7890335/Line.Select.zip)

supermerill, 2022-01-18:
it's because of the `over-brdge flow ratio`, if it's not 100%, I currently have to split the surface to apply the different flow.
Not optimal, but can't change that right now.

OverWorldD, 2022-01-29:
Ah, I see that makes sense. Thank you for explaining. I know now what it is and why it is there! And what to do about it :)