supermerill/SuperSlicer #2214: Monster Gap Fill

CCS86, 2022-01-13:
### What happened?

[image]

### Project file & How to reproduce

Layer 25, 26, 28, 61 has a gap fills that are .81mm, more than double the commanded width.

[MonsterGapFill.zip](https://github.com/supermerill/SuperSlicer/files/7859521/MonsterGapFill.zip)

### Version

2.3.57.9

### Operating system

Windows 10 x64

neophyl, 2022-01-13:
You have to set Print Settings>Perimeters & Shell>Quality>Overlapping External Perimeter>Also for all perimeters.  Make that ticked and slice again.

tome9111991, 2022-01-13:
I think similar to my issue https://github.com/supermerill/SuperSlicer/issues/1639

neophyl, 2022-01-13:
Yeah its exactly the same issue.  I've seen a few of these now and I asked awhile ago too which is why I know to set the Also for all perimeters box if you want to get rid of it with the current software.

CCS86, 2022-01-13:
Wow, much better, thanks guys!

CCS86, 2022-01-13:
Any drawbacks to leaving this setting enabled?

Mandragora-Arato, 2022-01-14:
Yes, the tooltip points a drawback. Flow isn't adjusted, example:
my extrusion width is 1mm, but my model wall thickness is 1,2mm. if i will set "Overlapping External Perimeter" printer will extrude 2mm width worth of material = terrible over extrusion.

CCS86, 2022-02-01:
@supermerill 

Is this slated for improvement, so we aren't forced to run overlapping perimeters in order to prevent unprintable gap fill widths?

supermerill, 2022-02-01:
I'm going to change overlapping by a threshold.
Maybe in the future, I may have the time to create a split algo to lay down two lines from the fat one.

CCS86, 2022-02-01:
Do overlapping perimeters conflict with the "enforce 100% fill volume" setting? Or, would they cause local over-extrusion (where they overlap), and a compensatory under-extrusion everywhere else from a flow rate reduction?

supermerill, 2022-02-01:
> Do overlapping perimeters conflict with the "enforce 100% fill volume" setting?

no

> Or, would they cause local over-extrusion (where they overlap), and a compensatory under-extrusion everywhere else from a flow rate reduction?

fat gapfill may be a bit on the extreme case, and so the extrusion position/ shape may be a bit inaccurate. But you can print at double the nozzle width, no problem. Imo the problem may lie more on the flow change than on the width.

> What is the reason "also for all perimeters" defaults to false?

the external ones can expand to the outside. For the inner one, they can't and so creates overextrusion vs fat gapfill.