supermerill/SuperSlicer #2216: vase mode incorrect handling of overhangs

sprior, 2022-01-13:
### What happened?

Starting with a working profile for solid objects I tried slicing the diamond tree model (https://www.thingiverse.com/thing:597501) switching to vase mode.

What I noticed in the print was that areas which are identified as overhangs were severely underextruded at a higher speed.

I'm wondering if there are other settings that should change automatically with vase mode, there is something wrong with the profile I've created, or there's a bug.

### Project file & How to reproduce

[DIAMOND_TREE.zip](https://github.com/supermerill/SuperSlicer/files/7865345/DIAMOND_TREE.zip)

### Version

2.3.57.9

### Operating system

Windows 10

### Printer model

Makergear M2 with a profile I've been setting up

neophyl, 2022-01-14:
Normally for slopes printed in vase mode you want your wall to be as thick as you can so that it can get support from the perimeter below.  I wouldn't be surprised at gaps between the layers in overhang areas with your 0.4 extrusion width.  Most vase prints I've seen and done use 0.7-0.8.. extrusion widths with a 0.4 nozzle to get a decent result.

Those area are showing as a higher volumetric flow rate in the preview.  While not a very high absolute value are you sure your extruder/nozzle combo can actually output it ?  I would expect a hotend to easily be able to handle 3.5mm/s but its best to rule out things.   If not then it would be under extruded.  Its honouring the volumetric flow set in your filament profile.  You can check using the preview, just change it from feature type to volumetric.

I also note that your bridges cooling is set to 100%.  As those areas are dark blue and so classed as bridges that is going to kick your fan up to 100% for those areas.  That with petg is going to reduce adhesion, leading to the possibility of it not sticking well to the perimeter below and combined with the overhang angle/shrinkage etc is going to accentuate any gaps as it drags filament over without it sticking.  
filament profile, then Id play with the 

So Id adjust the thresholds for  Overhangs under Print settings.  If you set 'Bridge speed and fan' to 75% instead of 50 it will not treat those areas as overhang anymore and so the fan shouldn't kick in at the higher speed.  Id also make the extrusion width thicker personally too.

sprior, 2022-01-14:
The 0.4 extrusion diameter shouldn't be an issue, I've printed this same model a bunch of times with PETG with that value using S3D, this is a test print for me getting familiar with SuperSlicer.  And aside from those areas marked as overhangs the print is perfect, and even those overhangs aren't much different in slop from the areas next to them.

Part of what I'm questioning is whether "bridge mode" should have been invoked at all, does bridging really even exist in anything you should be printing in vase mode?

I also notice the speed, I've got internal perimeter speed set to 80mm/sec and external set to 50% of that.  Bridge speed is set to 60mm/s, and those areas are being printed at 49mm/sec.  Since in vase mode ALL extrusions are external perimeters so I wouldn't expect the speed to ever exceed in this case 40mm/sec.

So my question is should whatever is causing these faces to be interpreted differently from the ones next to them have been turned off when entering vase mode?

neophyl, 2022-01-14:
I haven’t checked on the speed but the tagging of those faces is controlled by the overhang thresholds you have set. Why should that setting be turned off for vase ?  The setting is there, just set it to something that you want. At least you can, it’s a hidden fixed value in PS. With petg it’s more likely to be the cooling applied to those areas that is making the difference as I mentioned before. Have you actually tried printing it with the suggested settings changes to see if it prints better ? 

As for bridge, it’s not, it’s overhang and that does exist in vase mode. From what I recall it’s basically defined depending on how much of the perimeter is supported by the one below. Which is where the % comes in. It’s just that a lot of the same code is applied to both. Dont get hung up on labels.