supermerill/SuperSlicer #2217: Weird extrusion issue

robthide37, 2022-01-13:
### What happened?

During reg or vase mode on single wall prints their is an odd overextrusion where flow jumps to 20mm3 or drops to 0 almost like the slicer is trying to apply pressure advance. Its worse the smaller the layer height this shows it at .2
[image]

### Project file & How to reproduce

[pill_vase_hollow.zip](https://github.com/supermerill/SuperSlicer/files/7866097/pill_vase_hollow.zip)

### Version

2.3.57.9

### Operating system

win 10

### Printer model

troodon

supermerill, 2022-01-26:
I think the problem lies in the infinite resolution you use. As your model has some very very thin section, the extrusion is "rounded" to the lower or higher value.

robthide37, 2022-01-27:
Ok what resolution do you think is the point of diminishing returns? I am
using closed loop servos that are at 4000 cpr with 20 tooth pulley so
should I just stick to .01?

supermerill, 2022-01-27:
0.01 seems to work fine. I'll investigate a bit more when I'll have the time.