supermerill/SuperSlicer #2222: Travel speeds not converting from mm/s to mm/min in exported GCODE

linyongzheng, 2022-01-14:
### What happened?

Problem: Travel moves are 1/60th their intended speed because custom values are not being multiplied by 60 in exported gcode.

Expected: Feed rates should be multiplied properly.

### Project file & How to reproduce

[wasteBinLid.zip](https://github.com/supermerill/SuperSlicer/files/7867767/wasteBinLid.zip)

### Version

2.3.57.9

### Operating system

Windows 10

### Printer model

JGMaker Artist-D Standard

supermerill, 2022-01-15:
from the sliced gocde:
`G1 X81.25 Y81.25 F15000`
so it's the 250mm/s you entered.
Where is the problematic line?

linyongzheng, 2022-01-15:
Just resliced, still showing the error.

Lines 35-42 also showing only F250 instead of F15000
[image]
[image]
[image]

linyongzheng, 2022-01-15:
Looks like it is isolated to the custom gcode blocks where it is just directly along the values from the profile parameters but not converting proper units.
[image]
[image]

linyongzheng, 2022-01-15:
Looks like this is also prevalent in the prusaslicer 2.4.0. Also going to report issue there. Let me know if you want me to close it out here and have prusaslicer fix it.

linyongzheng, 2022-01-16:
According to Prusa, this behavior is "as intended" I think this is a poor excuse because it can easily be repeated without prior knowledge.

Regardless, since this is as intended, I'll close out this issue.

closed 2022-01-16 by linyongzheng

supermerill, 2022-01-16:
you can do the conversion yourself:
`G1 X-20 F{travel_speed*60}`
also you don't have to repeat the speed at each line, it's store until there is another F parameter