supermerill/SuperSlicer #2223: [Feature request]: Bridge through holes on mid-air flats

Miksu80, 2022-01-14:
It'd be nice to have an option to make bridge on holes that are on flat surface in mid-air. When doing prints like this:

[image]
[image]

You've got to have supports so they can be printed nicely:
[image]

But you can achieve a nice print also by bridging through holes like this, without a need for supports:
[image]

The bridges are easier to clean than supports and print quality is good this way

When deciding whether if its going to be a bridge or normal support, bridge length would be the ideal parameter -> if bridging length becomes too long, there might be sagging and it wouldn't work.

U can achieve this in CAD but it would be a lot easier and intuitive if it'd be a slicer option 😊

Miksu80, 2022-01-14:
Thank you, I totally missed this :)
Tested it and it works good although noticed a bug:
[image]
[image]

[Bridge_voids.zip](https://github.com/supermerill/SuperSlicer/files/7869065/Bridge_voids.zip)

neophyl, 2022-01-14:
Well it is experimental :)  Also while our human brains can look at that model and go , oh yeah that applies there but not there, its much harder for an algorithm.  Its doing what you asked in all parts of it.  I guess that's why the tooltip says to use a modifier to apply it in the areas you want.  That way you get the human discrimination into the loop.

Miksu80, 2022-01-15:
Was in a rush so didn't have time to examine it better, Ill close this one

closed 2022-01-15 by Miksu80