supermerill/SuperSlicer #2225: Add option to print part as an object

martijnschouten, 2022-01-14:
**Is your feature request related to a problem? Please describe.**
I work at a lab where we 3D print functional multi-material parts. We have a diabase H-series printer that can print both stiff an flexible materials. And I often print complex multi-material parts. We used to use Cura for this, but I starting trying to use Superslicer for this. This works really nice, especially since we now can just use macros and don't need a post-processor anymore.

However I now run in the problem that it appears to be impossible to have a multi-part object and have a object being printed as a object (with it's own bottom solid layers). It is also not possible to print multiple objects, because all objects are always dropped to the table.

To give an example, suppose I have two cubes on top of each other, one very flexible one very stiff
[image]

Then I want both part to be printed with solid top and bottom layers, as if they were two seperate object. But instead they are printed with infill at the interface.

[image]

**Describe the solution you'd like**
I think the best solution would be to have a modifier that allows parts to be printed as if they were an seperate object. Although just allowing objects to be printed at any height would also work (so having a option for disabling the auto drop to table feature as in Cura).

**Describe how it would work**
I guess there will be some property for every stl that says if it is an part or an object. If the modifier is applied it should be an object.

**Describe alternatives you've considered**
I now just print everything at 100 percent infill for the time being

supermerill, 2022-01-15:
> Although just allowing objects to be printed at any height would also work (so having a option for disabling the auto drop to table feature as in Cura).

not enough, as it will use a brdige flow as it's not on the build plate, so the first modifier is the better one i think.

here a hack for building in air:
[cube on top.zip](https://github.com/supermerill/SuperSlicer/files/7874941/cube.on.top.zip)

martijnschouten, 2022-01-15:
#not enough, as it will use a brdige flow as it's not on the build plate (but you can modify the flow to be similar to the normal one), so the first modifier is the better one i think.

That is exactly what I need! The first layer with a different material always is a difficult one, so it would be nice to be able to set the printing speed for that layer. Although a downside is of course that if there are any other bridges in the part, that they then also will be printed with that slow speed.

Thank you very much for the hacks. I'm not sure the hack of adding a layer of air between the parts will work in practice. I'm afraid the top cube won't stick to the bottom cube anymore.

I tried reproducing the other hack, but without success. What did you do there?
[cube on top MS.zip](https://github.com/supermerill/SuperSlicer/files/7875061/cube.on.top.MS.zip)

martijnschouten, 2022-01-15:
We found out what you did. In the second method, in the two-part object, the lower cube is set to be 0.1x0.1x0.1mm. This would indeed be a temporary hack that might work for us. Thanks.

martijnschouten, 2022-01-15:
I have found another hack to do this, which I think is a bit cleaner. In this hack, I make a separate part for the top layers of the bottom cube and bottom layers of the top cube. Then I set those two extra parts to 100 percent infill. It is a bit more work in the cad software, but for now it is a solution.

[cube on top MS2.zip](https://github.com/supermerill/SuperSlicer/files/7875310/cube.on.top.MS2.zip)

supermerill, 2022-01-16:
ah yes, full infill, nice idea

but you don't need to use parts for that, simply use a height range modifier
here is a project with it: (and i added a slowdown for the start of the second object)
[cube on top V3.zip](https://github.com/supermerill/SuperSlicer/files/7876209/cube.on.top.V3.zip)