supermerill/SuperSlicer #2229: Confirmation for color change or option to disable the feature?

elcojacobs, 2022-01-14:
**Is your feature request related to a problem? Please describe.**
I had 2 failed prints that had a layer shift before I noticed that it was caused by an accidentally added color change.

[image]

The problem was this gcode:
```
1 F7732.3307
G1 X222.281 Y189.708 E0.03926
; INIT stop printing object Integrated_Leadscrew_RF_Arm_V3-elco.stl id:0 copy 0
;LAYER_CHANGE
;Z:25.94
;HEIGHT:0.16
; custom gcode: before_layer_gcode
;BEFORE_LAYER_CHANGE
;25.94
; custom gcode end: before_layer_gcode

; stop printing object Integrated_Leadscrew_RF_Arm_V3-elco.stl id:0 copy 0
G1 Z25.94 F600
; custom gcode: layer_gcode
;AFTER_LAYER_CHANGE
;25.94
; custom gcode end: layer_gcode

; INIT printing object Integrated_Leadscrew_RF_Arm_V3-elco.stl id:0 copy 0
G92 E0
G1 E-0.6 F1200
G1 Z26.14 F600
; printing object Integrated_Leadscrew_RF_Arm_V3-elco.stl id:0 copy 0
G1 X215.691 Y189.108 F18000
G1 X215.179 Y190.345
G1 X215.91 Y191.386
G1 Z25.94 F600
G1 E0 F1200
;TYPE:Internal perimeter
;WIDTH:0.5343
G1 F9451.9072
```

It is way too easy to accidentally add a color change. If you just use the slider to inspect the gcode before printing, and accidentally click that + button right next to the slider, you have added a colour change.

[image]

If you miss that the top of the bar changed color, you just ruined your print before starting it.

[image]

**Describe the solution you'd like**
A confirmation dialog would be nice. 'Would you really like to add a color change at height 'x'? 
Or 'Please confirm the z-height of your color change': input field

**Describe alternatives you've considered**
Disabling color changes so the + button is not shown at all is also a good option. It just shouldn't be so easy to change color. If you have only one extruder, it's not something you do often.

neophyl, 2022-01-14:
Please don’t add more dialogs. I am not in support of the request. 

How can you miss that the slider changed colour?  Do you not check the preview after slicing ?  If you don’t, you should, it’s a good habit to get into. There’s a colour option, it’s very easy to check if there’s a colour change been applied. If you make use of the existing software capabilities. 

If you want to disable the colour change then edit the custom gcode section for your printer profile to remove the colour change command in there. Instant no more accidental colour change. Without effecting anyone else. You can always use the insert custom gcode option if you want to add one. That should make it safe enough for you.

elcojacobs, 2022-01-14:
Do you change color that often?
If you have your left hand on the enter key and the right on your mouse, it's click, press enter, and you have accepted the already filled in color change in the dialog.

I have added an update that there is indeed no need to confirm the height, just always show the color picker.

elcojacobs, 2022-01-14:
> How can you miss that the slider changed colour? Do you not check the preview after slicing ? If you don’t, you should, it’s a good habit to get into. There’s a colour option, it’s very easy to check if there’s a colour change been applied. If you make use of the existing software capabilities.

I always look at the gcode preview, but mostly have it in feature type mode. I sometimes use speed, fan speed, flow rate, layer time view. Whatever is relevant to me. I never have the gcode preview on 'color print', because that's never relevant to me.

neophyl, 2022-01-14:
One reason I am surprised is that when ever I add a colour change and then slice the preview window is automatically on the colour print option and NOT the feature option.  Its is very hard to miss that you have added a colour change. I have to change it back to Feature, which admittedly is a pain sometimes but it does mean that I have never accidentally added a colour without realising it. 

Does it not automatically change to that for you ?

I just tried it now to confirm my memory of what happens.

elcojacobs, 2022-01-15:
Does it still do that if you change to feature type, remove the color change by clicking the x, and then click on the + again?