supermerill/SuperSlicer #2235: Solid infill not the same as 100% infill

SharkWipf, 2022-01-16:
### What happened?

When setting infill to 100%, settings for solid infill are not applied to the infill structure.
As such, settings like ironing are not applied to the infill.

Setting Print Settings -> Infill -> Advanced -> Solid infill every x layers to 1 turns the entire infill solid, and ironing will be applied to the entire print.
There seems to be no way to apply ironing to the infill otherwise, as there is no "Ironing" pattern for the regular infill.
It would, IMO, be more sensible to generate all infill as solid infill by default when infill density is set to 100%.

### Project file & How to reproduce

[SidewinderX1-Idler-Lever-nosolid.3mf.zip](https://github.com/supermerill/SuperSlicer/files/7877410/SidewinderX1-Idler-Lever-nosolid.3mf.zip)

- Set infill to 100%
- Enable ironing for solid infills
- Observe the lack of ironing on solid infills
- Force all infills to be generated as solid infills by setting `infill_every_layers` to 1
- Observe ironing being applied to the now-solid infills, despite 100% infill also being solid.

### Version

2.3.57.9

### Operating system

Fedora Linux w/ AppImage

### Printer model

Sidewinder X2

supermerill, 2022-01-17:
just add 1000 bottom layers, no?

SharkWipf, 2022-01-17:
It's not that there's no workarounds available, as mentioned, setting `infill_every_layers` to 1 also works fine. 
It's more that it isn't intuitive/expected for 100% infill and solid infill to be treated differently while they are functionally the same. 
Changing all infill to solid infill at a 100% infill ratio would make more sense IMO.

supermerill, 2022-01-17:
But then, you can't discriminate the speed or the line width, if you need/want that.

SharkWipf, 2022-01-18:
Is that something that has to be discriminated though? Top and bottom can still be discriminated, it's just "100% infill" vs "solid infill", where the latter is, currently, pretty arbitrary if you want to specify any custom settings for it.  
The suggested workarounds are just that, workarounds, hacks to coax the program into doing what you want by using settings for things they're not meant for.  
Now I understand Slic3r and its forks aren't very KISS by design to begin with, and settings tend to tie in with each other in various ways, but I don't feel relying on confusing workarounds to get ironing on every infill layer is necessary here.  
Even if you want to keep the option of differentiating between solid infill (100% infill) and solid infill (solid infill), there are other options that don't rely on workarounds, like adding an "Ironing" setting to the regular infill type dropdown (ideally only selectable if at 100% infill), or even a separate option to "generate 100% infill as solid infill" after the "solid infill" settings.

supermerill, 2022-01-18:
There is already some safeguard for 100% infill, like int he pattern that can be used.
Maybe I should just forbid sparse infill higher than 90% ?

SharkWipf, 2022-01-18:
Hmm, that's an option too I guess, but then there are no "intentional" ways to get all-solid infill at all, only workarounds.  
It would also create the inverse of the issue you mentioned before, not being able to discriminate between the 2 "kinds" of solid infill, because like before, there'd only be 1 "kind" left, except it's the other kind this time.  

If going that route, making all solid infill the same seems like the more clear option from an UX standpoint to me, and when slicing it's automatically clear that this is happening from the resulting sliced print as well.  
It'd also be more understandable what's happening for new users coming from Slic3r/Prusa I think.

supermerill, 2022-01-18:
> It'd also be more understandable what's happening for new users coming from Slic3r/Prusa I think.

I'm currently the same as prusa: 100% infill is still taggued as sparse and not solid (brown vs  purple):

[image]

SharkWipf, 2022-01-19:
Ah, no, my bad, I was referring to differences between max 100% infill in Prusa and the proposed max 90% infill in SuperSlicer