supermerill/SuperSlicer #2236: Disable first layer/bottom support interface

Ypsilantea, 2022-01-16:
**Is your feature request related to a problem? Please describe.**
The denser, wider, interface layer below support towers is difficult to remove, often is broken up by other geometry, and can be unnecessary with good bed adhesion. As far as I can tell, there is no way to disable it. Also, dense interface layers are generated on the bottom of support towers printing off of other parts, which is unnecessary and often hard to remove.

**Describe the solution you'd like**
An option to disable first layer support interface and an option to have dense interface layers only apply on the top of support towers.

**Describe how it would work**
A check box in the "Options for support material and raft" submenu for first layer interface, and a check box in the "Options for support material interface" submenu to have interface layers only generate on the top of support towers.

**Describe alternatives you've considered**
The Hilbert Curve pattern option doesn't generate first layer interfaces, but also doesn't generate anything at all for the first layer.

supermerill, 2022-01-17:
For the moment, you can add a modifier for that first layer that reduce the bridge flow.

The ideal solution should be a way to control the bridge only for "bridges over supports" ?

Ypsilantea, 2022-01-17:
I'm not entirely sure we're talking about the same things here. This is what I'm talking about for the first layer support interface:
[image]

And this for interface layers on the bottom of support towers:
[image]

neophyl, 2022-01-17:
The dark green bottom interface layers when supports are generated on top of a part can be turned off in PS2.4.  Theres a setting called Bottom Interface Layers, which by default is set to "same as top" but can be set to a different value including 0.

From your pictures it looks like you are using an earlier version of Prusa Slicer.  The new bottom layer when printed on the bed is much more solid (a 90% fill by default) but can be set between 10-100%.  This can be changed too and you can also set the expansion value to make it smaller or larger.  Theres also a new snug support option.

supermerill, 2022-01-17:
Oh, yes, sorry.
The 2.4 is coming, i'm working mainly on that.

neophyl, 2022-02-02:
Even in 2.4 you cant completely get rid of the bottom layer.  You can change it between 10-100% and the default 90% makes it pretty much a solid sheet instead of the wide space which is an improvement,

You can however turn off the bottom dark green interface layers where the supports start on the model.  Just not the layer on the bed itself.

gaifeng8864, 2022-02-28:
> 我不完全确定我们在这里谈论的是同样的事情。这就是我说的第一层支持接口： [image]
> 
> 这适用于支撑塔底部的界面层： [image]

dear Ypsilantea, have you found a setting or method to remove Bottom interface layers? I am also deeply troubled by this problem. Bottom interface layers are very difficult to remove, it tends to destroy the bottom contact surface, I think this is unnecessary, at least there should be an option that can be freely chosen, for example, PrusaSlicer has a related setting.

gaifeng8864, 2022-02-28:
> I'm not entirely sure we're talking about the same things here. This is what I'm talking about for the first layer support interface: [image]
> 
> And this for interface layers on the bottom of support towers: [image]

dear Ypsilantea, have you found a setting or method to remove Bottom interface layers? I am also deeply troubled by this problem. Bottom interface layers are very difficult to remove, it tends to destroy the bottom contact surface, I think this is unnecessary, at least there should be an option that can be freely chosen, for example, PrusaSlicer has a related setting.

dfosantos, 2022-03-24:
I am indeed having the same issue with both top and bottom interface layers set to 0:

[image]
[image]

gaifeng8864, 2022-03-25:
> I am indeed having the same issue with both top and bottom interface layers set to 0:
> 
> [image] [image]

In which version was this done? I am using 2.3.57.11 now.

dfosantos, 2022-03-25:
I am using a nightly build (I had problems with the auto re-slice and this version solves it)
Build: SuperSlicer_2.4.58.2_2022-03-12

gaifeng8864, 2022-03-25:
> I am using a nightly build (I had problems with the auto re-slice and this version solves it) Build: SuperSlicer_2.4.58.2_2022-03-12

alright, thank you very much. I will try it out in the near future.

supermerill, 2022-03-26:
so it's solved?

dfosantos, 2022-03-26:
No. That nightly build only solves a different issue unrelated to this one. This interface layer problem remains

closed 2022-07-14 by supermerill