supermerill/SuperSlicer #2237: Weird patterns on flat top surfaces

Carib3D, 2022-01-17:
On flat top surfaces (not necessarily the upper surface) the pattern seems to be divided into irregular segments. This seems to apply to all patterns. On the example in the video I scroll up and down through 5 top layers. The top two solid layers display this irregular pattern, the next 2 down are okay and the lower one is irregular. Below this, just above the non-solid infill, there is a 'half filled' layer where the problem seems to start. 

The model is accurately flat, modelled in f-360, so It should have a flat top surface and pattern. I had this happen once before and I haven't been using SS that long so it's not just a one off thing. I haven't been able to replicate on a test piece however and with only two occurrences, I admit to not having much to go on. On this part I can't get rid of it and it is unsightly on the finished part.

[Top pattern.zip](https://github.com/supermerill/SuperSlicer/files/7884082/Top.pattern.zip)

supermerill, 2022-01-17:
you have to get all the stack (for 5 top):
 - top
 - solid
 - solid
 - solid over-bridge (if != 100% flow)
 - internal bridge

Carib3D, 2022-01-17:
> you have to get all the stack (for 5 top):
> 
> * top
> * solid
> * solid
> * solid over-bridge (if != 100% flow)
> * internal bridge
> 
> surfaces can merge only when they are the same type.

Sorry, I don't follow what you mean, can you elaborate?
BTW this model slices okay in Prusa Slicer.

supermerill, 2022-01-17:
please save your project file (crtl+s) and share it here, so I can reproduce.

Carib3D, 2022-01-17:
Sorry, I thought they were in the initial zipped file. Here is second attempt
[Guide main.zip](https://github.com/supermerill/SuperSlicer/files/7884235/Guide.main.zip)

supermerill, 2022-01-17:
Your top is just in-between the two layers, so it's just where the slicer tries to cut, so it depends on other things like normals or don't know.
Prusa has the same problem:

[image]

If PS 2.4 doesn't, then it will fixed with SuSi2.4