supermerill/SuperSlicer #2239: Support Calibration

SphaeroX, 2022-01-18:
SuperSlicer has a lot of cool calibration tests, I would like to extend that to calibrate the support better, currently I do it manually.

I have attached the plate allocation.

Primarily it's about the spacing, but manually you can test other options afterwards.

[image]

[support_test.zip](https://github.com/supermerill/SuperSlicer/files/7890597/support_test.zip)

supermerill, 2022-01-18:
Thanks for the model.
But ~1 hour is a bit long. I wonder if it's needed to be that tall. Also, is the top volume need to be that big?
I think I can cut the printing time a bit more by removing the very top solid fill.

Also, can you explain how to use the results?

The project file only change the top distance, But I guess the model can also calibrate for the bottom one?

Also I'm wondering if it's possible to create a "tower", so each object can be printed on top of each other. Maybe a bit too difficult to have a reliable result on the top one....

SphaeroX, 2022-01-18:
No problem, I could optimise the whole thing and write a proper manual for it if you can implement it.

That was just a basic idea, because I just got to grips with it. Give me a little time and I will prepare everything :)

SphaeroX, 2022-01-18:
I also like the idea of the tower, but unfortunately you can't influence the support via the height range modifier?

supermerill, 2022-01-18:
> I also like the idea of the tower, but unfortunately, you can't influence the support via the height range modifier?

yes, true. So no tower.

What do you think of that?
[support_test_totry.stl.zip](https://github.com/supermerill/SuperSlicer/files/7891946/support_test_totry.stl.zip)

[image]

6 minutes to print, test of flat surface, bot & bottom, unconnected island, slope.
But maybe it's too packed and not wide enough to be relevant...

SphaeroX, 2022-01-21:
Actually, the object is quite good, only the place where you have the support at the bottom and top needs to be improved.

The spots are too thin, because the support is completely connected to the other one, the upper part tears, as you can see in the picture...

[image]

[image]

The second point is that I would also like to have a radius upwards (overhang), so you can adjust the angle well and XY distance from the object.

I'm sorry I don't have much time at the moment but trust me, I'll keep at it and I'm happy if I can support them.

SphaeroX, 2022-01-22:
I would prefer this test, why?
You can validate the result better, you can put a screwdriver and push the support material away with the help of the lever. If it then goes away well and the component looks good, the test is OK. What do you think?

Then I would make a calibration process for it and describe it with the help of pictures and text what you have to pay attention to.

[Uploading support test v3.zip…]()

SphaeroX, 2022-01-22:
Here you can check how well you can get the support off the top component and what the component looks like.

[image]

Here you check whether you can get it off well from below and whether it sticks to the XY axle.

[image]

Finally, you can also look at the overhang to see if it is close enough to the support.

[image]