supermerill/SuperSlicer #2241: Sacrificial layer not bridging properly

paulrmatei, 2022-01-18:
### What happened?

Sacrificial layer isn't bridging properly
Attached a screenshot from superslicer, one from prusaslicer(2.4)
Prusaslicer
[image]

Superslicer
[image]

### Project file & How to reproduce

Import the following file
[connector-v2.zip](https://github.com/supermerill/SuperSlicer/files/7891684/connector-v2.zip)
Set layer height to 0.16 or below
Go to layer ~109 (~11.7mm)

### Version

2.3.57.9

### Operating system

SuperSlicer-ubuntu_18.04-gtk3-2.3.57.8

supermerill, 2022-01-18:
remove "only one peri on top" at least for this layer.
That's something I have to improve, yes.

Duplicate of 1494

paulrmatei, 2022-01-20:
Thank you
That worked

closed 2022-01-20 by paulrmatei