supermerill/SuperSlicer #2242: Gap Fill issue

Jurgis-R, 2022-01-19:
Hi,
I have a couple of questions, first may sound dumb, but  I cant seem to find a MAX width limiter for gap fill (in mm or % of nozzle diam), see the attached picture and 3mf file.

2question : Can SS later versions than v2.3.569 (in expert mode of course) have a fully configurable travel, lift, retract ACC+Jerk? 
Example1 => set travel ACC, travel Jerk till print point - 0.5mm X/Y (configurable with possibility 0mm) DCC to print type Jerk, print. 
Example2 if Retract on + Lift Z on => Example1 + line print end, set Retraction ACC 8000mm/s^2, Retraction Jerk 6mm, make retract, adjust Lift ACC and Jerk then lift .
GCODE would look like 
M204 S8000; (set retract ACC, Klipper needs S parameter)
M205 E18; (set retract jerk)
G92 E0 ; reset extrusion distance
G0 E-6 F36000; (retract)
M204 S6000; (set lift ACC, Klipper needs S parameter)
M205 Z4; (set lift jerk)
G0 Z+0.3 F18000; (lift)

3question: is there a possibility to make retract and lift simultaneously? (use min ACC and min Jerk taken from Retract and lift parameters)
GCODE would look
M204 S6000; (set lift ACC, Klipper needs S parameter)
M205 Z4 E4; (set lift jerk)
G92 E0 ; reset extrusion distance
G0 E-6 Z+0.3 F18000; (retract + lift)

I use these M204 and M205 and F numbers on actual prints, they are not made up. My printer is Anycubic I3 Mega on skr 1.3 +tmc2130 on X/Y, A4988 on Z and E + Klipper. If You are wondering about high 300mm/s Z feedrate, just know that You can go way higher even with A4988 and 0.7A and 8mm pitch thread rods, tested 600mm/s but you get mechanical knocking from fast up/down, sticking with 300mm/s for smoothness. 

4question: any plans to separate sparse infill angle + increment from Solid/Top/Bottom angle + increment, just like in Cura?
If used rectilinear/monotonic for Solid/Top/Bottom the benefit is much better surface quality and better Solid/Top/Bottom lines adhesion with perimeter lines. Example Solid/Top/Bottom angle=15, increment 22.5. So in every 4 layers you will get 0 to 90 degree fill pattern. Also sometimes you position a part that you want sparse infill lines to go in specific angle, separate from Solid/Top/Bottom angle.

5question/suggestion about Z seam: if the lift +retract simultaneously idea works good, maybe it can be made for lowering head+ deretracting to start the print line? Example G1 Z0.7 E+6. Or even better variable configuration like Z seam length=0.2 mm steps=10.
GCODE example
M204 S6000; (set ACC of Lowering head + deretraction + seam print) Klipper needs S parameter
M205 X5 Y5 Z4 E6; (set jerk)
G1 X10 Z1 E6
G1 X9.98 Z0.97 E5.4
G1 X9.96 Z0.94 E4.8
G1 X9.94 Z0.91 E4.2
G1 X9.92 Z0.88 E3.6
G1 X9.9 Z0.85 E3
G1 X9.88 Z0.82 E2.4
G1 X9.86 Z0.79 E1.8
G1 X9.84 Z0.76 E1.2
G1 X9.82 Z0.73 E0.6
G1 X9.8 Z0.7 E0

6question: how to make ArcWelder work with arguments in SS? I`m dumb again and did not find anything useful using google
I`m have only basic knowledge of computer code. If I use just d:\ArcWelder.exe in post-processing scrips, everything works, but if I use d:\ArcWelder.exe --mm-per-arc-segment=1.0 --min-arc-segments=24 --max-gcode-length=70 --extrusion-rate-variance-percent=0.025 --default-e-precision=6 --default-xyz-precision=4 --allow-dynamic-precision --allow-travel-arcs --path-tolerance-percent=0.02 -r=0.025 or even one argument it does not work :((

No problem with some longer GCODE even marlin and SKR 1.3 LPC1768 100MHz, 31.97KB RAM, have enough power to print just from USB with 256bufered lines and baud 250k(compiles with 85% used ram Configuration_adv.h in zip if questions arise)

Currently  I`m sticking with SS v2.3.569 just because later slicer versions GCODE with Klipper have weird travel behavior and noise.

[Bondtech.zip](https://github.com/supermerill/SuperSlicer/files/7894647/Bondtech.zip)

neophyl, 2022-01-19:
For question 1 then please read this thread https://github.com/supermerill/SuperSlicer/issues/2214

Cant answer the rest as they are feature requests mainly.  You might want to read other posts here and in the discussion section though as I know there has been some regarding z lift, retraction etc and some of your question may have been covered in those.

Jurgis-R, 2022-01-19:
Thanks neophyl for the answer to question1, in #2214 they solve it but not 100% correctly, because if You check "overlapping external perimeter" + "Also for all perimeter" you have a possible over-extrusion which I, and I think all the rest of printer enthusiasts do not need..

Jurgis-R, 2022-01-19:
Uploaded a picture for clarification
[image]

supermerill, 2022-01-24:
2: no jerk settings. you may want to create your own gcode writer script.
3: not at the same time. That can create spikes.
4: maybe... it's more of a gui issue, as duplicating a set of settings for all types of fill is a bit verbose.
5: the unretract has to build pressure, and i fear that you won't be able to do that without being at the right Z. Maybe a part of the unretract can be done in the unlift. A bit complicated to modify.
6: https://github.com/FormerLurker/ArcWelderLib#running-arcwelder-from-slic3r-slic3rpe-prusaslicer-and-superslicer

Jurgis-R, 2022-01-25:
supermerill, 
In my 1st question how to limit gapfill max width neophyl replayed to look at #2214 and that I need to check "Overlapping External Perimeter" + "Also for all perimeters" so I solve gapfill max widh, but does it not also enable overextrusion of external and internal perimeters?? I do not want overextrusion enabled.. Or do I missunderstand "Overlapping External Perimeter" + "Also for all perimeters"??

2."you may want to create your own gcode writer script" => I could make my own Jerk and ACC, but SS (checked 2.3.57.9) does not have all working "extrusion_role`s" that I can put in "between extrusion roles" field in custom gcode 
working roles: SolidInfill/InternalInfill/TopSolidInfill/BridgeInfill/Perimeter/ExternalPerimeter/Skirt/Brim/ 
not working: OverhangPerimeter/GapFill/Retract/Travel or I`m writing it wrong??
{elsif extrusion_role=~/.*OverhangPerimeter*./}M204 S8002 ; not working ??
SET_VELOCITY_LIMIT SQUARE_CORNER_VELOCITY=3 ACCEL_TO_DECEL=6500 ; Gapfill Jerk + Acc
{elsif extrusion_role=~/.*GapFill*./}M204 S8002 ; not working ??
SET_VELOCITY_LIMIT SQUARE_CORNER_VELOCITY=5 ACCEL_TO_DECEL=6500 ; Gapfill Jerk + Acc
{elsif extrusion_role=~/.*Retract*./}M204 S30002 ; not working ??
SET_VELOCITY_LIMIT SQUARE_CORNER_VELOCITY=12 ACCEL_TO_DECEL=100000 ; Retract Jerk + Acc
{elsif extrusion_role=~/.*Travel*./}M204 S8002 ; not working ??
SET_VELOCITY_LIMIT SQUARE_CORNER_VELOCITY=15 ACCEL_TO_DECEL=6500 ; Retract Jerk + Acc
{endif}
didn`t test: SupportMaterial/SupportMaterialInterface/WipeTower/Wipe/Wipe_extra/Wipe_extra_perimeter/Extra_Wipe/ 
Else I do not know how to write my own code/program/script, if I could I would and not ask questions.. 

3."not at the same time. That can create spikes." create how? on low/med/high speed jerk/acc?? I never had spikes but maybe 
you can get Spikes on slow speed like retraction=50mm/s and lift =50mm/s?? 
I dont get any spikes at retraction=300mm/s and lift =300mm/s , and for example, that I have tested => retraction and lift speed 300m/s with jerk 5 ACC3000 is completely different compared with the "same" 300mm/s speed with jerk12 and ACC8000. I use 300mm/s jerk12 and ACC8000, will test faster speeds just for retraction when i get my TMC2226, now I have A4988 (MAX only 0.9A) my stepper can take 2.3A (https://www.aliexpress.com/item/32798466987.html?spm=a2g0o.9042311.0.0.27424c4dnETZlL). This lets me print big parts in PLA with 27mm/^3 on 280C on volcano hotend with 0.8mm nozzle (H0.3, W1.5mm) with ~70mm/s speed and I get very good layer adhesion.
My travel speed is: X/Y400mm/s, Z300 Jerk5 ACC8000 and my personal conclusion is: faster retractions + faster travel=better surface, no blobs, spikes ect.. Tested only PLA/ABS/PETG
In short: only real life test will show if spikes are made or not with 300mm/s Jerk5 ACC8000, but I cannot test retract +lift simultaneously, because I cant write my own GCODE, that`s why slicers are for.. 
Also I am 100% certain, that You will get different results even compared to the same federate, but with different acc/jerk, like F18000, but ACC2000 Jerk3 and F18000, but ACC3000 Jerk5 and F18000, but ACC8000 Jerk12.

4.Sorry English is not my native language, maybe I wrote something complicated, I meant "can You separate sparse infill from the rest?" I thought =>just add another line for SPARSE infill like this
[image]

5."the unretract has to build pressure" As I understand GCODE of the printing process, even theoretically I see no pressure problems, because all new lines START to get deposited in a space of +1exact layer heigh above the previous layer, please correct me if I`m wrong.. 
Lets take en actual example from SS GCODE
[gcode.zip](https://github.com/supermerill/SuperSlicer/files/7935848/gcode.zip) 
As I understand every line has its EXACT order: a)Retract b)Lift (amount from settings) c)Move to next print line start point d)Restore layer Z (now the nozzle is in free space above the previously printed line height exactly +1layer height from previous layer, because this is an unprinted line and in order to print it, the nozzle needs space = layer height/mm to deposit the molten plastic and so the printer prints the next line above the previously printed line) e)Deretract f)Start to print.
from GCODE at random: from line 27723 to 27736 line we have
a)G10 ; retract
b)G1 Z8.39 F18000 ; lift Z
c0)G1 X121.6093 Y127.8711 F24000 ; move to first External perimeter point (acceleration)
..
cn)G1 X136.6937 Y96.8091 ; move to first External perimeter point (deceleration)
d)G1 Z8.09 F18000 ; restore layer Z
e)G11 ; unretract
G1 F4613.3706 change feedrate
f)G1 X135.8363 Y97.6685 E0.125939 ; External perimeter

So again, theory is one thing, but actual printing of lowering head+ deretracting to start the print line is another and, of course, I also can`t test real print, because I can`t code the slicer..

6.Thanks after a lot of testing made "d:\ArcWelder.exe --mm-per-arc-segment=1.0 --min-arc-segments=36 --path-tolerance-percent=0.02 --resolution-mm=0.025" to work, but if I add "--max-gcode-length=70" or "--allow-travel-arcs" then it just outputs normal SS  GCODE, if I add --allow-dynamic-precision --default-xyz-precision=4 --default-e-precision=6 then SS gives en error. I guess it`s my problem not knowing how to write a simple command line script with arguments... Also I cant`t test a script if I am not 100% certain that I wrote it correctly..

supermerill, 2022-01-25:
next time, create 6 issues, it's a chore to answer.

1) choose your sin

2) It's not implemented, but planned for the future.

3) 
[image]

4) and for the top, the bottom. Like for the connection, the pattern.

5) As I said, maybe a part of the unretract can be done in the unlift. But if it's entirely done before it's at the right z, the filament will be pressurized in the nozzle before it's able to extrude and will ooze puch more.

6) I'm not the author of the ArcWelder script.

Jurgis-R, 2022-01-26:
supermerill Thank You for the answer
I`m new to github and next time will, create a separate message for a separate question.

closed 2022-01-26 by Jurgis-R