supermerill/SuperSlicer #2244: Cannot change top infill in Vase mode.

rglory, 2022-01-20:
### What happened?

In Vase mode in particular or just in regular mode when top solid layers is set to 0 Top Infill is grayed out, but it still affects printing. It is a problem with extruders that jam on Ironing and it difficult to disable it, as you need to turn vase mode off, increase top solid levels, then change to infill to something else and then enable vase mode again.

### Project file & How to reproduce

Any, just turn on Vase mode or Top Solid set to 0

### Operating system

ubuntu 21.10

supermerill, 2022-01-26:
can you send a project file where you have the issue (the top pattern is grey out and printed)
If I activate the spiral vase, I can still modify it.

And a picture as I'm not sure of the setting you talk about.

rglory, 2022-01-26:
It works the same with any project, just set "Solid Layers:" "Top:" to 0 (Vase mode does that implicitly) then this parameter is grayed:
[image]
The problem it still affects how top of base printed and does ironing if you do not turn it off.

supermerill, 2022-01-26:
[image]

rglory, 2022-01-26:
I am not sure what the difference, for me both on Linux and Windows they grayed and I cannot modify them. I run version 2.3.56

martinal22, 2022-07-19:
I have this same problem on 2.3.56 on macOS. If vase mode is selected, the top infill is greyed out and stuck on the previous setting. I have to turn vase mode off, then change the top infill, then turn vase mode back on. I was coming here to open in issue but found this one already existed. I'll try to play with other settings later to see if I can find anything else out.

supermerill, 2022-07-19:
share your porject, can't replicate.

martinal22, 2022-07-19:
[vase_mode_project.3mf.zip](https://github.com/supermerill/SuperSlicer/files/9143414/vase_mode_project.3mf.zip)

Is this what would help? I did File -> Save Project while in vase mode and the top infill was greyed out.

supermerill, 2022-07-19:
Can't replicate on 2.3.57.12 nor 2.4.58.3, windows and ubuntu22.04

Also, some settings aren't compatible with spiral vase, seems like you're on an old version.

martinal22, 2022-07-19:
I'll update tonight and try again. I'm on 2.3.56 macOS.

martinal22, 2022-07-20:
I updated to 2.3.57 and it works. Sorry about that. Thanks for the replies and the amazing tool.