supermerill/SuperSlicer #2245: Extruder and bed temperatures not added to gcode

elfren, 2022-01-20:
### What happened?

Gcodes M109 and M190 are not added to gcode:

;TYPE:Custom
; custom gcode: start_gcode
G28 ; home all axes
G1 Z5 F5000 ; lift nozzle
; custom gcode end: start_gcode
; custom gcode: start_filament_gcode
; Filament gcode
; custom gcode end: start_filament_gcode
M107 ; disable fan
;LAYER_CHANGE
;Z:0.2
;HEIGHT:0.2
G1 Z0.2 F7800 ; move to next layer (0)

Expected behaviour:
These two lines to be included at start of gcode:
M140 S128 ; set bed temperature
M104 S250 ; set temperature

Note: This started after editing Printer Settings/Custom G-code/End G-code to add the following two lines which were in my PrusaSlicer settings:
M104 S0 ; turn off temperature
M140 S0 ; turn off be
[ZLimitMount_A.zip](https://github.com/supermerill/SuperSlicer/files/7903325/ZLimitMount_A.zip)
d

### Project file & How to reproduce

File/Export fails with Extrusion multiplier.  It's set to 2.05.  This setting has worked on my delta for many years.

### Version

Version 2.3.57.8

### Operating system

Windows 10

### Printer model

Custom delta running Smoothieware

elfren, 2022-01-20:
These two lines should have been written to the gcode file:
M190 S128 ; set bed temperature and wait for it to be reached
M104 S250 ; set temperature

neophyl, 2022-01-20:
Your config has the option " Only custom start g-Code" ticked under Printer Settings>Custom G-Code. (tick box right at the top)

That means nothing is automatically added like it was previously.  Before if you didn't manually add in the correct start gcode then it would try and helpfully add them for you.  Lots of people complained about that so the tickbox to make it only add what you put in there was added.  
Your profile has it enabled so naturally its not adding anything for you.

If you want that option enabled then you need to add something like :- 
M140 S[first_layer_bed_temperature] ; set bed final temp
M104 S[first_layer_temperature] ; set extruder final temp
M109 S[first_layer_temperature] ; wait for extruder final temp
M190 S[first_layer_bed_temperature] ; wait for bed final temp

Note that Prusa Slicer does not have the tickbox and will always add stuff (there are open requests on their github to change that behaviour too).  

While SuSi and PS are closely tied together there are differences and you MUST double check the profiles when trying to convert one to the other.  Its up to the user to verify their custom profiles.

elfren, 2022-01-20:
Thanks. That has it sorted.

closed 2022-01-20 by elfren