supermerill/SuperSlicer #2246: Ironing Pattern Calibration does not fit onto the AnyCubic Mega S preset

jodrell2000, 2022-01-20:
### What happened?

When generating the Ironing Pattern Calibration SuperSlicer gives me an error "Error: A tool path outside the print area was detected"

### Project file & How to reproduce

- load the Anycubic i3 Mega S printer preset
- Chose the "Ironing pattern calibration" from  the Calibration Menu
- Click the "Over-Bridge calibration" button

### Version

2.3.57.9

### Operating system

MacOS 10.15.7

### Printer model

Anycubic i3 Mega S

jodrell2000, 2022-01-20:
I worked around this by modifying the printer preset to have a 200x200 bed, then generating the Calibration print, then changing the bed size back to the correct 210x210

supermerill, 2022-01-21:
If you click on the "arrange" button, will it move the pieces again out of the platter?
If yes, I have to add a kind of safety offset.

closed 2022-04-01 by supermerill