supermerill/SuperSlicer #2247: CPU Usage at 100% for large slice operation

arne182, 2022-01-20:
### What happened?

When slicing a large file. Superslicer uses all the resources it can get. with my 6 Core 12 Thread machine it could at least leave me some resources so that I could do normal work. It is nice that it is multithreading but maybe I could suggest if there is 12 threads available use 10. Leaving 2 for the OS to be able to run smoothly while the slicing is occuring. If there is only 4 threads on the system, let it use 3 to process the slicing.
[image]

### Project file & How to reproduce

Slice at 0.1 layer height over a 400mm length model.

### Version

2.3.57.9

### Operating system

Windows 10

### Printer model

Ratrig V Core 3

supermerill, 2022-01-20:
I can add a preference item to let you limit the number of threads.

arne182, 2022-01-20:
Now that sounds like a good idea. Something like threads or percent where it defaults to 80%?