supermerill/SuperSlicer #2249: Internal Bridge Infill bug - Only half of the Infill is created.

NVNDO, 2022-01-20:
### What happened?

The Internat Bridge Infill is missing half of its structure.
I expected a base layer of Infill that support the next layer completely, but it only support half of it, creating an uneven base for upcoming layers resulting in gabs. 
[image]
[image]
[image]
[image]

### Project file & How to reproduce

[Bridge Infill Bug.zip](https://github.com/supermerill/SuperSlicer/files/7907525/Bridge.Infill.Bug.zip)

No steps needed to recreate. Just scroll down to layer 29

### Version

2.3.57

### Operating system

Windows 10

### Printer model

Ender 3 Pro

supermerill, 2022-01-20:
Can you share your project file so i can reproduce?
[image]

NVNDO, 2022-01-20:
Hey there, yes of course.
Just added the original F360 file and a 3mf.
Hope that works to reproduce

So quick… did not expect that.

Let me know! Thx a lot and best regards
Nando

[image]<https://user-images.githubusercontent.com/6536403/150400697-ca5409dd-1bde-4267-9a41-f993d00a390e.png>

NVNDO, 2022-01-20:
Did the files work? Could you reproduce the error?

perkki123, 2022-01-20:
Hi,
 Had same issue. Problem occured when selected either: Keep only bridges or Keep bridges an overhangs in Advanced/No perimeters on bridge area.

NVNDO, 2022-01-20:
> Hi, Had same issue. Problem occured when selected either: Keep only bridges or Keep bridges an overhangs in Advanced/No perimeters on bridge area.

I have "No perimeters on bridge areas" disabled. I just tried all other options in case it would make a difference, but it doesn't. The error persists.

supermerill, 2022-01-21:
> Did the files work? Could you reproduce the error?

no, I don't have the 3.mf
Please  do File->save porject as
then zip it and drop it here

NVNDO, 2022-01-21:
Hopefully now.
For me all previous Zips and files have worked..

Please do File->save project as

NVNDO, 2022-01-21:
[Cube Error.zip](https://github.com/supermerill/SuperSlicer/files/7912702/Cube.Error.zip)

NVNDO, 2022-01-24:
Any news..? No pressure... but as this error is ruining my prints I will (at least at this point) keep on using other alternatives.
With best regards.

supermerill, 2022-01-26:
can't reproduce on 2.3.57.9
[image]

supermerill, 2022-01-26:
I need your project file.
when you have your bug, click on fil-> save project as
then zip it and drop the zip here.

NVNDO, 2022-01-26:
You resized the cube, that's why you can't reproduce.
If leaving the cube as it is 20x20x20mm the error appears, regardless of what infill.

It's clear to see in the image that your cube is much bigger than mine.

[image]<https://user-images.githubusercontent.com/6536403/151207494-aa81e9b5-857c-43fc-bb36-63d8e68dbf6e.png>

supermerill, 2022-01-26:
The project file contains your settings. I need them to reproduce the issue.
The .3mf you added is not a project file saved by superslicer as it doesn't contain any setting.

NVNDO, 2022-01-26:
[Cube_Error_SSr.zip](https://github.com/supermerill/SuperSlicer/files/7945678/Cube_Error_SSr.zip)

Last try...

supermerill, 2022-01-26:
It's an stl, not a project file.
Do you need a video to show you how to save a project and share it?

NVNDO, 2022-01-27:
[Cube_Error_27.01.21.zip](https://github.com/supermerill/SuperSlicer/files/7950072/Cube_Error_27.01.21.zip)

Its exactly what I had previously send via email that apparently did not arrive in this chat. 
If that doesn't reproduce I'll just let it be.

NVNDO, 2022-01-27:
Nice! Sorry for taking so long to get the correct file send.

closed 2022-04-01 by supermerill