supermerill/SuperSlicer #2250: Variable for start g-code for initial tool

Merlin246, 2022-01-20:
Using an E3D toolchanger.

When printing, the first tool used in the print is not necessarily T0. However I cannot find a way to specify the first tool in the start gcode (if I don't specify the tool, no tool is picked up at all as it is not added in with the default start gcode). This may be from my own ignorance of a specific variable that does this that exists, if so please forgive me, if not:

I would like a variable to be added something like "initial_extruder" or use the currently existing "next_extruder" (found in custom gcode under printer settings > tool change gcode) to be supported in the start gcode script so I don't have to use T0 and then potentially have it instantly change tools. It's a small thing that doesn't have a huge impact but it will just make it more streamlined.

My start gcode if curious:
;Start gcode for E3D Toolchanger

M140 S[first_layer_bed_temperature]; set bed to temperature

M568 P0 A1; set T0 to standby
M568 P1 A1; set T1 to standby
;M568 P2 A1; set T2 to standby → not yet built
;M568 P3 A1; set T3 to standby → not yet built

;Homing
G28 C ; home c
G28 Y ; home y
G28 X;  home x

;Homing Z
M116; wait for all temperatures → allows bed more heatsoak time and instant printing after probing Z instead of waiting for hotends
G28 Z; home z

G29 S1; Activate mesh compensation

T0; → ideally T[next_extruder] or similar

supermerill, 2022-01-21:
check the tooltip:
initial_extruder or initial_tool or current_extruder

Merlin246, 2022-01-21:
Brilliant cheers