supermerill/SuperSlicer #2251: Building on macOS fails with OpenVDB error.

ornamentist, 2022-01-21:
### What happened?

After binary downloads to macOS failed I tried building from source on an m1 mac running macOS 12.1.

After running the macOS build script this went well until this error:

```
Found OpenVDB: /opt/homebrew/include (found suitable version "9.0.0", minimum required is "5.0") found components: openvdb
-- OpenVDB ABI Version: 9
CMake Error at cmake/modules/FindOpenVDB.cmake:322 (message):
  IlmBase::Half can not be found!
Call Stack (most recent call first):
  cmake/modules/FindOpenVDB.cmake:346 (just_fail)
  CMakeLists.txt:472 (find_package)

-- Configuring incomplete, errors occurred!
See also "/Users/stu/tmp/SuperSlicer/build/CMakeFiles/CMakeOutput.log".
See also "/Users/stu/tmp/SuperSlicer/build/CMakeFiles/CMakeError.log".
done
[8/9] Building Slic3r...make: *** No rule to make target `Slic3r'.  Stop.
make: *** No rule to make target `gettext_po_to_mo'.  Stop.
done
ls ROOT
Build.PL			SECURITY.md			sandboxes
BuildLinux.sh			build				src
BuildMacOS.sh			cmake				t
CMakeLists.txt			create_release.py		tests
LICENSE				deps				version.inc
PrusaSlicer.mo			doc				xs
PrusaSlicer.sublime-project	lib
README.md			resources
ls ROOT/build
CMakeCache.txt	CMakeFiles
ls -al ROOT/build/src
ls: /Users/stu/tmp/SuperSlicer/build/src: No such file or directory
chmod: /Users/stu/tmp/SuperSlicer/build/src/BuildMacOSImage.sh: No such file or directory
./BuildMacOS.sh: line 156: /Users/stu/tmp/SuperSlicer/build/src/BuildMacOSImage.sh: No such file or directory
ls -al ROOT/build
total 64
drwxr-xr-x   4 stu  staff    128 21 Jan 18:31 .
drwxr-xr-x  29 stu  staff    928 21 Jan 18:30 ..
-rw-r--r--   1 stu  staff  32411 21 Jan 18:31 CMakeCache.txt
drwxr-xr-x   7 stu  staff    224 21 Jan 18:29 CMakeFiles
```

I have OpenVDB and OpenEXR installed via Homebrew so I'm not sure what the root cause is here. Could be related to https://github.com/prusa3d/PrusaSlicer/issues/4445 ? 

### Project file & How to reproduce

[error.zip](https://github.com/supermerill/SuperSlicer/files/7911257/error.zip)

### Version

latest clone

### Operating system

macOS 12.1

### Printer model

none yet

ornamentist, 2022-01-21:
I also tried installing `ilmbase` via Homebrew, which didn't help.

supermerill, 2022-01-21:
building master?

ornamentist, 2022-01-21:
A straight git clone of the repository, so I guess the default branch?

metacollin, 2022-03-03:
@ornamentist IlmBase is a keg-only formula, which means Homebrew doesn't put it in the normally searched paths.  

If you run `export PKG_CONFIG_PATH="/usr/local/opt/ilmbase/lib/pkgconfig"` after installing it via homebrew, then rerun cmake, it should find it without issue.

ornamentist, 2022-03-06:
Good catch, thanks.

Using Fish shell on an M1 mac with macOS 12.x, I did:

```
brew install ilmbase
set -gx LDFLAGS "-L/opt/homebrew/opt/ilmbase/lib"
set -gx CPPFLAGS "-I/opt/homebrew/opt/ilmbase/include"
set -gx PKG_CONFIG_PATH "/opt/homebrew/opt/ilmbase/lib/pkgconfig"
./BuildMacOS.sh
```

(Note the /opt base for Homebrew files on 12.x). Which got through the build process until:

```
chmod: /Users/stu/tmp/SuperSlicer/build/src/BuildMacOSImage.sh: No such file or directory
./BuildMacOS.sh: line 158: /Users/stu/tmp/SuperSlicer/build/src/BuildMacOSImage.sh: No such file or directory
ls -al ROOT/build
total 80
drwxr-xr-x   5 stu  staff    160  6 Mar 14:53 .
drwxr-xr-x  29 stu  staff    928  6 Mar 14:52 ..
-rw-r--r--   1 stu  staff  39393  6 Mar 14:46 CMakeCache.txt
drwxr-xr-x   7 stu  staff    224  6 Mar 14:37 CMakeFiles
drwxr-xr-x  18 stu  staff    576  6 Mar 14:46 src
```

ornamentist, 2022-03-06:
I looked at the contents of `https://github.com/supermerill/SuperSlicer/blob/master/src/platform/osx/BuildMacOSImage.sh.in` to try and use it, but it needs variables expanded first?

safademirhan9, 2025-01-17:
Any updates for this issue? 

`brew install ilmbase` also doesn't work:
`Error: ilmbase has been disabled because it is not supported upstream! It was disabled on 2024-02-07.`