supermerill/SuperSlicer #2254: Ensure vertical thickness causes occasionally solid infill mid object & doesn't work with 3 perimeter and more

rovili, 2022-01-21:
### What happened?

Solid infills are created mid object when "ensure vertical thickness" is activated.
It does not happen with the newest and older PrusaSlicer versions.

Super Slicer - solid infill example
[image]

Prusa Slicer - solid infill example
[image]

Super Slicer - 3 perimeter thickness/overhang-support:
[image]

Prusa Slicer - 3 perimeter thickness/overhang-support:
[image]

### Project file & How to reproduce

[moai-small-no_overhang.3mf.zip](https://github.com/supermerill/SuperSlicer/files/7915460/moai-small-no_overhang.3mf.zip)
[3perimeter_moai-small-no_overhang.3mf.zip](https://github.com/supermerill/SuperSlicer/files/7915496/3perimeter_moai-small-no_overhang.3mf.zip)

### Version

2.3.57.9

### Operating system

macOS 12.2

supermerill, 2022-01-24:
Can you try without the Horizontal shell ->"no soilid infill over ... perimeters" option?

rovili, 2022-01-24:
[image]

[image]

supermerill, 2022-01-24:
Superslicer with 0 for "no solid infill over...":
[image]
seems the same as ps.
Is it okay for you?

rovili, 2022-01-24:
You're right, with 0 it helps for the overhang and it is similar/same as in PS. I agree.

But my picture with solid infill from above was from today with 0 - it happens only occasionally.

acwest, 2022-02-09:
I am getting similar mysterious solid infill layers, in version 2.3.57.10. The `ensure vertical thickness` and `no solid infill over` settings do not seem to affect it, in my case. In the screenshot below, you can see that some layers are marked as having an overhang. At those layer heights, a complete top solid infill is inserted below the layer, and a complete bottom solid infill is inserted above it. Another thing that is interesting is that the central sun gear that is showing these problems does not have the overhangs or extra layers when sliced further from the planet gears. Those are probably two separate issues, but I don't really believe in coincidence...

[image]