supermerill/SuperSlicer #2256: Editing in G-Code boxes unable to utilize any keys aside from the alphanumeric

hieppo, 2022-01-21:
### What happened?

When editing the G-Code, i.e. Start and End, under Printer preset, The only keys that can use are the alphanumeric keys. None of the standard keys like Enter, Backspace, Cursors can be used. Hence, new lines can't be added. Characters can't be deleted. The only thing can be done to delete is to highlight the area where changes need to be done and overwrite with new entries.

On another note, the content of the G-Code boxes are found the config file. But for every new line needs a /n to signified a new line. Why does it needs to display this in the config file?

### Project file & How to reproduce

1. Run GTK2 or GTK3 appimage both same results
2. Click on Printer Settings
3. Select Custom G-code
4. Try to add new lines in Start G-code box but fail to do so
5. Try to delete word but fail as well

### Version

2.3.57.9

### Operating system

Ubuntu 18.04

### Printer model

Creality Enter 3 Pro

hieppo, 2022-01-21:
The same symptons happen with PrusaSlicer as well. Both slicer shows the same issue.

supermerill, 2022-01-21:
> the content of the G-Code boxes are found the config file. But for every new line needs a /n to signified a new line. Why does it needs to display this in the config file?

'\n' is a common key to write "newline" As every setting need to written in one line, it's the way to record multi-line strings.

For the key problem I'll check with my ubuntu vm when i'll have the time.

supermerill, 2024-05-21:
can't reproduce on ubuntu 22.04
@hieppo is it still a bug on latest stable for you?