supermerill/SuperSlicer #2257: Top layer with 1 perimeter below top surface

rovili, 2022-01-21:
### What happened?

Objects with lower top layers get 1 perimeter option, even if something is on top. External perimeter should go through all height.

1 perimeter layer:
[image]

above again multiple perimeters:
[image]

### Project file & How to reproduce

[example.3mf.zip](https://github.com/supermerill/SuperSlicer/files/7916357/example.3mf.zip)

### Version

2.3.57.9

### Operating system

macOS 12.2

rovili, 2022-01-21:
Maybe a misunderstanding from my side. But I think this should only be applied to the last layer? Here it happens in mid of the object.

supermerill, 2022-01-24:
All red one are top surfaces. A top surface is a surface without anything on top of it.
If you want that only on the topmost layer, you can use a modifier to modify the perimeter count directly.

rovili, 2022-01-24:
I understand - we can close this ticket. =)

closed 2022-01-24 by rovili