supermerill/SuperSlicer #2262: Bridging at set angle for object regardless of viability

HopsDa, 2022-01-22:
### What happened?

There should be no overlap of the bridging lines and so the bridge should be in the correct direction to support upper layers. I have cut half the model off to have a second object and there the bridging is correct. sorry for the bad explanation but i think you will understand from the screen shots. 

Thank you!

### Project file & How to reproduce

[image]
[config.ini.zip](https://github.com/supermerill/SuperSlicer/files/7918966/config.ini.zip)

### Version

2.3.57.9

### Operating system

mac with M1 pro

### Printer model

RatRig 400mm, octopus v1.1, raspi4

neophyl, 2022-01-23:
Looks like your attached project didn’t make it to the issue. I’m just seeing text and not a downloadable link.

HopsDa, 2022-01-24:
[Bridging not working.3mf.zip](https://github.com/supermerill/SuperSlicer/files/7923284/Bridging.not.working.3mf.zip)

neophyl, 2022-01-24:
As far as the algorithm is concerned it is a single bridged area.   As either direction is good for one and not for the other it picks one.  
You have the Advanced option "No perimeters on bridge areas" set to keep only bridges.  So its removing all the perimeters. that would normally  support  the edges of the bridge when its printed in the short direction.

The section where you have excised for test purposes only travels one way and so the calculation is different. and it puts it in the best way for that area.

If you set 'No perimeters on bridge areas' to Disabled  or  Keep bridges and overhangs then it should print better.  If you really really dont want perimeters on those sections then you will have to use modifiers to manually set the angles of the bridging in those areas.

HopsDa, 2022-01-24:
it does indeed make a difference and i haven't thought about modifiers at all. about time to give them a try!

Thank you!

closed 2022-01-24 by HopsDa