supermerill/SuperSlicer #2264: Thin walls creates artifacts "in the air" - no connection to layer before

MaGe-1, 2022-01-22:
### What happened?

- tried to print temperature calibration tower with 0.6 Nozzle
- Fine detailes are deleted, see picture
[image]
[image]

### Project file & How to reproduce

See steps before

### Version

tried with most recent and last recent stable release

### Operating system

windows

MaGe-1, 2022-01-22:
Prusaslicer 2.4 does the same 
[image]

neophyl, 2022-01-23:
That’s because the thin wall algorithm is flawed and has always been.  Plenty of existing issues about it already on the Prusa GitHub. 

Just don’t use it. Make sure that your walls are all printable, tweak your extrusion widths if needed.  It’s one setting that I always turn off.

supermerill, 2022-01-24:
can you put here your saved project file (to have the settings used)
also, what is exactly the problematic part for you (the pointy thing?)

MaGe-1, 2022-01-27:
The issues is, that there are several  locations, where the material drop because its floating in the air
its getting more and more and causes the print to fai
[image]

the end of the data musst be .3mf
I changed it to be able to upload it
l

supermerill, 2022-01-27:
> I changed it to be able to upload it

I don't see the file

MaGe-1, 2022-01-27:
now the file only
![grafik](https://user-images.githubusercontent.com/85059294/151445387-793c22be-baca
[image]
-414f-9aaa-b2cffaca6b8e.png)
 
its a stl. did it work?

supermerill, 2022-01-27:
that is what I see: dead link and garbage text.
How did you do that?
[image]

MaGe-1, 2022-01-30:
[Temperature calibration.zip](https://github.com/supermerill/SuperSlicer/files/7966936/Temperature.calibration.zip)

I honestly dont know. It should work this time...

supermerill, 2022-01-30:
With that project, here is what I have:
[image]
Left: with (default) overlapping perimeters ; Right: without

Can't reproduce your picture you post on first message. Can you?