supermerill/SuperSlicer #2265: ![](https://cdn.discordapp.com/attachments/824333729127333929/934521259851665449/unknown.png)

VikBormann, 2022-01-22:
### Version
2.3.57.9

### Operating system type + version
Windows 10 (build 19043), 64-bit edition

### Behavior
I have already made some prints with the slicer on the BLV Cube. Everything has always worked wonderfully. Now I have a large lithophane. The slicer stopped calculating at Z 177.9mm and the print was cancelled. Unfortunately, the part is 190mm.

[image]

neophyl, 2022-01-23:
6 million lines of gcode ?  That seems rather excessive. 
please attach a zipped up project file as requested when opening a new issue so that the devs can take a look at this unusual issue. Without it the chances of working out what’s wrong are much reduced.

VikBormann, 2022-01-25:
Hello

Is here a file size limit?
I can't upload my file.
The zip. file is 28MB

Jurgis-R, 2022-01-25:
Hi, 
neophyl, if You are working with large stl`s and want at least of 0.025mm resolution GCODE, that has more life like shapes, like a lot of changing radius-es You easily get above 1x10^7 of standart GCODE lines (not ARC optimized..) 
I tested with my model (height is 85mm) on Win7 x64b +8GB RAM +16GB virtual memory on SS 2.5.3.57.9, slicer does not break at 3.6+x10^7 GCODE lines, takes about 1.5h long to slice.. , also I have another 200mm height model, that I have sliced that gives 27+x10^7 GCODE lines, but only after I added 16GB of virtual memory on C disk. 
see pic and file
[image]
[slif ratas mod 1 20CM 2mm sienele perpiestas PILNAVIDURIS.zip](https://github.com/supermerill/SuperSlicer/files/7937598/slif.ratas.mod.1.20CM.2mm.sienele.perpiestas.PILNAVIDURIS.zip)

VikBormann, maybe You also need a lot of RAM and maybe to increase virtual memory to at least 16GB??

I can`t kill SS slicer just by slicing 1GB+ STL`s, but I constantly crash it even with small STL`s when doing a lot of settings readjustment + slicing, like 30-50 in a row, can`t reproduce exactly. Now I just constantly save.. :)