supermerill/SuperSlicer #2266: Model after slicing out of printing Area

Wingnut7489, 2022-01-24:
### What happened?

As you can see on the pictures. Every time I slice a imported STL file. It is shown outside the print area.

I am using the latest version for Mac.
[image]
[image]

### Project file & How to reproduce

[image]
[image]

### Version

2.3.57.9

### Operating system

MacOS Monterey 12.1 

### Printer model

Weedo Tina 2

mtaumike, 2022-01-25:
May want to edit your report as both pictures are the same from what I can tell.

supermerill, 2022-01-25:
> As you can see in the pictures, every time I slice an imported STL file it is shown outside the print area.

Sorry, I don't see that. Please explain it to me like if I'm a 10yo.