supermerill/SuperSlicer #2267: Not slicing model properly when "External Perimeter First" is activated

WickedInvi, 2022-01-24:
### What happened?

Activated External perimeter. Crashed multiple times when it doesn't the inside outer perimeter does not slice. Tested with the previous version and slices fine. Copied the settings to the backup SS folder I keep.

Here is the model
[image]
On the current version
[image]
On version 2.3.57.8
[image]

### Project file & How to reproduce

[config.zip](https://github.com/supermerill/SuperSlicer/files/7928868/config.zip)

### Version

2.3.57.9

### Operating system

Windows 10

### Printer model

Ender 3 with SKR mini E3 v2, Klipper, Dragon Hot end, Belted Dual Z

supermerill, 2022-01-25:
can't reproduce the problem
[image]

Can you share your project file?

WickedInvi, 2022-01-26:
[Calibration cube.zip](https://github.com/supermerill/SuperSlicer/files/7940175/Calibration.cube.zip)

closed 2022-04-01 by supermerill