supermerill/SuperSlicer #2268: Ubuntu ARM64 Build failing

tobyworks, 2022-01-25:
Would be nice to have support for building SuperSlicer under a Ubuntu ARM 64 architecture. So we can do some server-side slicing it is supported in PrusaSlicer i believe.

supermerill, 2022-01-25:
I should have a raspi3 laying somewhere....
But it's really not my priority right now.
If someone can debug the way to build it on the platform, I can merge the changes.

aegean-odyssey, 2022-01-31:
I've successfully built SuperSlicer as a Debian (.deb) Bullseye package for both arm64 and amd64. You can find the [repository, here](https://github.com/aegean-odyssey/superslicer). I tried to follow Debian's packaging guidelines so that maybe someday SuperSlicer can be picked up an official way by Debian maintainers. The provided GitHub workflow should generate both the amd64 and arm64 packages, but GitHub's arm64 runner is too slow to complete the build within GitHub's 6-hour time limit.

Perhaps there is something here that can help you with your Ubuntu build.

troypiggo, 2022-03-14:
G'day AO - awesome work on the deb arm64 build thankyou.  I'm having a little trouble installing it.  I'm running Ubuntu 20.04 LTS on RPi4 arm64, and some of the depends can't be met:
```
The following packages have unmet dependencies:
 superslicer : Depends: libboost-chrono1.74.0 (>= 1.74.0) but it is not installable
               Depends: libboost-filesystem1.74.0 (>= 1.74.0) but it is not installable
               Depends: libboost-locale1.74.0 (>= 1.74.0) but it is not installable
               Depends: libboost-log1.74.0 (>= 1.74.0) but it is not installable
               Depends: libboost-regex1.74.0-icu67 but it is not installable
               Depends: libboost-thread1.74.0 (>= 1.74.0) but it is not installable
               Depends: libilmbase25 (>= 2.5.4) but it is not installable
               Depends: libopenvdb7.1 but it is not installable
               Depends: libtbb2 (>= 2020.3) but it is not going to be installed
               Depends: libwxbase3.0-0v5 (>= 3.0.5.1+dfsg) but it is not going to be installed
               Depends: libwxgtk3.0-gtk3-0v5 (>= 3.0.5.1+dfsg) but it is not going to be installed
E: Unable to correct problems, you have held broken packages.
```

20.04 seems to only have libboost-chrono1.71.0 and so on.  Any tips or pointers in the right direction?  Happy to (try to) build my own if you could briefly give an overview.

aegean-odyssey, 2022-03-20:
Yes, differences in the shared libraries between Debian and Ubuntu. Building the package specifically for focal (20.4) is probably your best bet. I'm not familiar with building packages for ubuntu focal directly on the RPi4, but I _think_ the same build process should work:

```sh
sudo apt-get update -q --yes
sudo apt-get install --no-install-recommends --yes build-essential git-buildpackage equivs

git clone https://github.com/aegean-odyssey/superslicer.git
cd superslicer

sudo mk-build-deps -i -t 'apt-get --no-install-recommends --yes'
sudo rm -f superslicer-build-deps_*
gbp buildpackage -uc -us -J2 -j2 --git-submodules --git-upstream-tree=SLOPPY
```