supermerill/SuperSlicer #2269: Feature Request/Client certificate Authentication

Pieterv24, 2022-01-25:
**Is your feature request related to a problem? Please describe.**
I run my printer interfaces on the public web, however to protect them I make use of 2-way SSL.
What this means is that instead of a username or password, a user has to authenticate to my server with a Client Certificate.
SuperSlicer is unable to upload gcode files to Moonraker/Octoprint due to this.

**Describe the solution you'd like**
I'd like the ability to specify and add a client certificate in the phisical printer dialog as option besides pasword/username or api-key.

**Describe how it would work**
Looking at the current codebase I found that libcurl is used for doing over network request to hosts like Octoprint/Moonraker. Adding support for this feature would require to add an option to libcurl, found [here](https://curl.se/libcurl/c/CURLOPT_SSLCERT.html).
This option points to a file (P12 or PEM) and add that to the libcurl execution.  

In order to achieve this, a new option entry in `PhysicalPrinterDialog.cpp` is needed to select the client certificate and logic should be added to `Http.cpp` and possibly `Octoprint.cpp` and other physical printer types to use this option and append it to the curl request to the host.

**Describe alternatives you've considered**
None, In my opinion a client cert is one of the most secure ways to secure my hosts.

**Additional context**
I am currently looking into how to implement this myself, Any help, tips or remarks are welcomed

mtaumike, 2022-01-25:
So, do you have a working server side SSL client authentication working?

Moonraker, by default, doesn't use an API key though Octoprint does.

I'd think that adding client authentication to SuperSlicer is going to require that the backend server side is going to require work as well and would be beyond the scope of this project.

Pieterv24, 2022-01-25:
> 

I am currently exposing my moonraker/mainsail hosts from my local network though an reverse proxy NGINX server configured to only allow connections with a valid Client Certificate.
Currently I can only access my Mainsail interface through my browser and upload though that, but I'm unable to send any sliced gcode files directly to the host using SuperSlicer.

I don't think it should have to be extended past the scope of "SuperSlicer is able to pass a client cert to host (regardless of host) if the user wants to do so".

It might not have the largest subsection of user maybe, but I think it would be a great addition for people who want to use it (including me).

Pieterv24, 2023-11-27:
I believe this issue can be closed as the feature has been added

closed 2023-11-27 by Pieterv24