supermerill/SuperSlicer #2271: Wipe Tower have one very slow line

Siwat2545, 2022-01-26:
### What happened?

[image]
The print is using the Palette 2S with P2PP, however, the issues seem to be from the slicer.
The nozzle moves stupidly slow over these lines and increases print time drastically.

### Project file & How to reproduce

[AnimalCell.zip](https://github.com/supermerill/SuperSlicer/files/7938516/AnimalCell.zip)
https://github.com/tomvandeneede/p2pp

### Version

2.3.57.9

### Operating system

Windows 10

### Printer model

Prusa i3 MK3S with Palette 2

Siwat2545, 2022-01-26:
Seems to have something to do with ramming, but the p2pp manual tells me to disable it.
However when I enable ramming, this happens
[image]

Are there any ways to set a constant wipe speed?

supermerill, 2022-01-26:
I added a limit on speed from filament_max_speed, but i forgot to multiply by 60 [shy]

Siwat2545, 2022-01-27:
If I don't set the filament_max_speed, P2PP just yeet the nozzle left and right over the wipe tower, haha
[image]
But no worries, I'll use canvas while I want patiently for your wonderful bug fix :)

closed 2022-02-19 by Siwat2545