supermerill/SuperSlicer #2273: Print temp

emtcharlie, 2022-01-26:
### Version
_Version of SuperSlicer used goes here (help->about)_
2.3.57

### Operating system type + version
_What OS are you using, and state any version #s_
Win10

### Behavior
* _Describe the problem_
After slicing with nozzle temp of 240c and transferring file to CR-10s Pro V2 via SD card the nozzle temp defaults to 150c. Only tested with PETG profile
* _Steps needed to reproduce the problem_
Consistent
* _Expected Results_
Nozzle temp should match settings in slice.
* _Actual Results_
Nozzle temp is set at 150c vs the 240c expected.
  * _Screenshots if useful_

_Is this a new feature request?_
Related guides for writing feature requests: http://meta.stackexchange.com/a/259196 http://nickohrn.com/2013/09/write-great-feature-request-bug-report/

#### Project File (.3MF) where problem occurs
_*Upload a Project File* (.3MF) `Plater -> Save Project as...`_
_You have to zip it (or rename it to .zip) to be able to drag& drop it here_

neophyl, 2022-01-26:
Your printer profile isnt set up for a 2 stage warmup is it ?

If you would please attach a zipped up project file as requested when opening an issue we can load it and check all your settings to confirm if this is the case.

A 2 stage warmup is designed to warm up to a no ooze temperature first, then do things like mesh levelling and then after all that is done finally heat up to the print temperature.  For example the Prusa Mini is configured to use 170 degrees.  There are lots of 'reports' of faulty mini printers on the Prusa forum and github when the behaviour is by design.

Edit - I have just added a CR10s Pro V2 config from the wizard and it does indeed have a M105 S150 command in the startup gcode so your printer and the slicer is doing exactly what it is supposed to do.
M104 S150 ; set temporary nozzle temp to prevent oozing during homing and auto bed leveling

emtcharlie, 2022-01-27:
Thank you for replying. Please disregard this ticket. I'm a moron lol

closed 2022-01-27 by supermerill