supermerill/SuperSlicer #2276: Gcode Meta commands Duet3D / conditional Gcode ERROR

jackantubis, 2022-01-27:
### What happened?

This Gcode is indicate as en error in SuperSlicer during slicing

if !move.axes[0].homed || !move.axes[1].homed || !move.axes[2].homed || !move.axes[3].homed || !move.axes[4].homed
	G28

It's a kind of programming Gcode with variables and condition made in RRF by Duet3D: 
https://duet3d.dozuki.com/Wiki/GCode_Meta_Commands

The use of this particular Gcode is to home the printer only if the homing is not already done. 
Could be used as a macro or start/end Gcode.

THanks for your amazing job ;-)

### Project file & How to reproduce

Place this code in the "start Gcode": 

if !move.axes[0].homed || !move.axes[1].homed || !move.axes[2].homed || !move.axes[3].homed || !move.axes[4].homed
	G28

Load a STL and try to slice, an error occur.

### Version

2.3.56.6

### Operating system

windows

### Printer model

JUBILEE

neophyl, 2022-01-27:
I'm curious what happens if you are using an earlier version of RRF before they added meta command and it gets those lines ?

I suspect that the RRF generator in SuSi can't parse those commands as it doesn't have support for them.  One way to 'fix' would be to add something similar to a comment or enclosing tag to indicate to SuSi that the contents of the tag should just be added verbatim to the output gcode.
That way you would have to add a new gcode flavour, could just use the current RRF one and the tagged system could work with any of them to add in custom lines of gcode in future.  

Of course I may be talking utter rubbish :)  Best wait for Merill.

supermerill, 2022-01-27:
I'm going to resolve this by two ways:
You will be able to add a `{ignore_legacy(true)}` to disable the legacy format using the `[]`
```
{ignore_legacy(true)}
if !move.axes[0].homed || !move.axes[1].homed || !move.axes[2].homed || !move.axes[3].homed || !move.axes[4].homed
G28
```
And you will be able to escape `[` (and the `{`), is you want to still use the legacy format elsewhere (it's still very often used in current configs) : 
```
if !move.axes\[0].homed || !move.axes\[1].homed || !move.axes\[2].homed || !move.axes\[3].homed || !move.axes\[4].homed
G28
```

jackantubis, 2022-01-27:
Thanks I'll try as soon the version is out 👍

MarkXYZ-Mtl, 2022-02-14:
Hi, 
Is this fix is in the 2.3.57.11 version?

I'm trying to get this macro call to RRF 3.3 in plain text: "G1 E{-global.TC_Retract} F750"

{ignore_legacy(true)}
G1 E{-global.TC_Retract} F750
{ignore_legacy(false)}
->this will cause error popup

G1 E\{-global.TC_Retract} F750
->this cause every variable after this one in the tool change custom gcode in plain text

Thanks 
Marc

supermerill, 2022-02-14:
yes, it should.
if it doesn't work, try  `G1 E\{-global.TC_Retract} F750`

MarkXYZ-Mtl, 2022-02-15:
Hi,
It worked!!

In my second example I had a typo, I swear I've tried the "\" before "{" and every other variables in the tool change g code was in plain text....  Maybe I did a brainfart..... The only thing I can see is that I had opened version 2.3.57.6 first then 2.3.57.10, and 2.3.57.11 on my PC. 

Thanks for the help
And this slicer is awesome!

closed 2022-07-14 by supermerill