supermerill/SuperSlicer #2279: Missing spacing values in the settings comments appended at the end of an exported gcode

AheadWind, 2022-01-27:
### Version
_Version 2.3.57.3

### Operating system type + version
_Windows 10 20H2

### Behavior
* In the appended settings comments at the end of an exported gcode file, all the values involving line spacing say the variable name and then have no associated value. 
* Going into expert mode, I can see that the spacing values are all filled in with mm values that are consistent with their associated overlap percent. (these values are correct)
* expected results
; extrusion_multiplier = 1,1
; extrusion_spacing = 0.6449
; extrusion_width = 0.72
...
; external_perimeter_cut_corners = 0%
; external_perimeter_extrusion_spacing = 0.
; external_perimeter_extrusion_width = 0.72
* actual result:
; extrusion_multiplier = 1,1
; extrusion_spacing = 
; extrusion_width = 0.72
...
; external_perimeter_cut_corners = 0%
; external_perimeter_extrusion_spacing = 0.6824
; external_perimeter_extrusion_width = 0.72
  * _Screenshots if useful_
[image]

#### Project File (.3MF) where problem occurs
[Y1 Reduced - Jacob.zip](https://github.com/supermerill/SuperSlicer/files/7953868/Y1.Reduced.-.Jacob.zip)

supermerill, 2022-01-27:
spacings are not active, the labels are grey. that's why

AheadWind, 2022-01-27:
Thank you!  Does that mean that spacings are not being used by the main slicing engine? and if so then is the graphic in the extrusion width-spacing box not accurate? - Im asking since I am trying to determine real line width for my parts.

supermerill, 2022-01-27:
That means they are deactivated. If you set one, then the corresponding width will be deactivated and so won't have a value in the config file.

In the voodoo magic of the slicer, the spacing isn't used directly, but the computed width from it is.