supermerill/SuperSlicer #2281: Single Layer, with an Extra Layer on Top?

CCS86, 2022-01-28:
### What happened?

This one is pretty odd:

[image]

### Project file & How to reproduce

[DoubleLayer.3mf.txt](https://github.com/supermerill/SuperSlicer/files/7959961/DoubleLayer.3mf.txt)

### Version

2.3.57.9

### Operating system

Windows 10 x64

CCS86, 2022-01-28:
These additional print moves don't show up in the sliced preview, only in the gcode preview.

CCS86, 2022-01-28:
I think I may have gotten rid of these segments by disabling the variable layer height feature and then reenabling it.

supermerill, 2022-01-28:
seems like a goce viewer issue.
I'll see if it's still present in the 2.4

CCS86, 2022-01-28:
I was wondering whether the gcode viewer was just displaying part of the next layer. I did not try to print.

I changed many print settings and the behavior persisted. Once I changed the minimum segment length and disabled variable layer height, the behavior went away and didn't return when I reverted those settings.

supermerill, 2022-02-09:
Not present in 2.4.
I won't bother fix that in the 2.3 branch.

CCS86, 2022-02-09:
Sounds good, thanks! I just downloaded the latest 2.4 nightly to poke around. The live gcode viewer is pretty awesome.

closed 2022-07-14 by supermerill