supermerill/SuperSlicer #2282: Feature Request: introduce print host upload hostname/IP as variable in post-process scripts

MrDix, 2022-01-28:
If you want to use accurate print time estimations in gcode you can use the klipper_estimator script:
https://github.com/Annex-Engineering/klipper_estimator#post-process-mode

This would mean you have to add a line like this in the field 'Post-process scripts' in SuperSlicers:
C:\bin\klipper_estimator.exe  --config_moonraker_url http://192.168.0.50 post-process 

For an accurate print time estimation this plugin considers all of the printer side configuration (velocity, acceleration, activated pressure advance and input shaper) and needs to fetch this information from the concrete printer the gcode gets uploaded to.
Unfortunately this means you insert the IP address you've configured in your 'Printer Setup' -> 'Physical Printer' section to your 'Print Settings'. Afterwards this configured 'Print Settings' can not be used for an other printer anymore, as the print time estimation would use the wrong data and result in a wrong estimation.

Having the information provided in the 'Printer Setting# -> 'Physical Printer' ->'Print Host upload' ->  'Hostname, IP or URL' field as a post-process environment variable, would allow us to use this variable for the actual klipper_estimator call and would keep the 'Print Settings' configuration universal for several printers.

LazeMSS, 2022-07-24:
I think this is the same as: https://github.com/supermerill/SuperSlicer/issues/2862

supermerill, 2022-07-24:
In next nightly, the variables from physical printer will be available as environment variables, like for the other ones.

MrDix, 2022-07-24:
Great to hear! Thank you very much <3

closed 2024-02-11 by supermerill