supermerill/SuperSlicer #2286: no decelleration before some corners but not others

kevenwyld, 2022-01-30:
### What happened?

When printing this box (included below) my printer is decelerating into every corner radius except one where the speed through the corner is constant.  This is causing some issues with the printer as it can't handle the rapid change in direction and has some effect on the surface quality.

I don't seem to be able to find the deceleration at all in the gcode/speed preview though and I'm not even sure where it's configured at all.  Is it possible that speed changes within layers are not shown in the preview?

I ended up redesigning the part to avoid this and accidentally overwrote the original project file.  I do have the gcode generated from the print though which I have included.  I've also included two pictures below which show a "good" and "bad" corner.  When observing the print the nozzle/head clearly slow down for all corners except for this one particular inside one.  The part seam is aligned but not in the location where this issue occurs.

My questions are:

1. Where is this deceleration configured?
2. How can I see the decelleration in the gcode preview?
3. Is this a bug and is there a way to avoid missing this deceleration step on a single corner?

[image]
[image]

### Project file & How to reproduce

[misc-partbox-body-mesh_0.14mm_PLA_.gcode.zip](https://github.com/supermerill/SuperSlicer/files/7967100/misc-partbox-body-mesh_0.14mm_PLA_.gcode.zip)

As mentioned earlier I overwrote the original file and modified the part.  This is the gcode generated by superslicer though.

I can attempt to recreate the problem again with a simpler part if that would be helpful.

### Version

2.3.57

### Operating system

archlinux

### Printer model

Railcore II (duet w/ reprapfirmware gcode flavor)

supermerill, 2022-01-30:
in print -> speed, there is a little section with some acceleration control.
If let to 0 (disabled), the gcode shouldn't contain any accel/decel command, so everything is up to your firmware.
It seems you have only the travel acceleration set, try to disable it.

kevenwyld, 2022-01-30:
Thanks, I have all of the accelerations disabled in that section with `Default` set to 0.  It appears that `Travel` has a default value of 1500 though, I will try to set this to 0 without adjusting the other acceleration types.

[image]

I've included a project file from the current print.  I'm running another test now and will see if the problem persists.  After this one I will try one with the acceleration changes you suggested.

[misc-partbox-body-mesh.3mf.zip](https://github.com/supermerill/SuperSlicer/files/7967153/misc-partbox-body-mesh.3mf.zip)

kevenwyld, 2022-01-30:
Here's a video of the current print from the project I uploaded previously.  You can hear and see that there is a noticeable deceleration on all the corner radii except for the inside near right corner (so lower right of the screen if that makes sense).

https://user-images.githubusercontent.com/11936535/151722489-317e283c-b586-490f-9117-9dbae34a9b89.mp4

It's sounding like this is an issue with my firmware configuration though if the deceleration isn't part of superslicer so I'll start digging around in there to see if I can find the issue as well.  Thanks for the help.

kevenwyld, 2022-01-31:
I've confirmed this is a firmware configuration issue.  Specifically I had (reprap) M566 [...] E speed change set very very low by accident.  I'm guessing that caused the feed rate to slow automatically around the corners.  I'm still not sure why exactly, maybe a bug, or just that the feed rate is calculated to be within the limits of the extruder settings, but it has nothing to do with superslicer so I'm closing this.

closed 2022-01-31 by kevenwyld

kevenwyld, 2022-02-03:
So just a follow up here.  It turns out that this was not entirely firmware.  I had set the `resolution` parameter to `0` instead of the default `0.0125`.  I thought this wouldn't cause any harm since most of the parts I print are tiny, and to be fair it didn't cause any harm for those parts, but when it came time to print these relatively large trays at high speed reprap couldn't handle the extra resolution and got confused during high speed direction changes when coming around these corner radii.  Anyway this isn't a bug so I'll leave the issue closed, but if someone runs into this pay attention to the warning in the tooltip for `resolution` and leave it default.