supermerill/SuperSlicer #2288: wxWidgets GetItem Index out of Range

CCS86, 2022-01-31:
### What happened?

SuperSlicer was idle in the background and through this error:

[image]

### Project file & How to reproduce

[wxwidget.3mf.txt](https://github.com/supermerill/SuperSlicer/files/7973404/wxwidget.3mf.txt)

### Version

2.3.57.9

### Operating system

Windows 10 x64

supermerill, 2022-02-09:
Can't reproduce. May be hard to diagnose.
Can you make this error message appear reliably? If yes, can you explain the steps to make it appear? Or I can give you a debug build that will output a more useful message.

I think this kind of things may appear when the screen change (like going from a 100% scale display to a 150% one), or something like that.

feketeimre, 2022-08-10:
This happens every time i open the superslicer-gcodeviewer.exe. Although with a slightly different message:
[image]

Might be unrelated but my hunch says otherwise: If i press continue im unable to load files. Gives the following error: The selected file ... does not contain a valid gcode.

supermerill, 2022-08-10:
still here?

feketeimre, 2022-08-10:
> still here?
Had the same problem with gcode viewer. But i realized it was an old release, 2.3.57.11 so i deleted my post.
Im sorry, i didnt meant to bump the issue.

supermerill, 2022-08-10:
add a new message if it happens again in new versions.

closed 2022-08-10 by supermerill

CCS86, 2022-10-29:
2.5.59.0

[image]