supermerill/SuperSlicer #2289: Bed and Hot End not heating before starting print

jfkerekes, 2022-01-31:
### What happened?

So I am just getting started with SuperSlicer and I generated my first print. The printer I am using uses Klipper firmware. Version of SuperSlicer is 2.3.57

So I generated the Gcode and started the print. Upon starting the print it immediately homed and tried to start printing without warming up the bed and hot end and halted the print due to low E0 and Bed temp... I pre-heated the bed and then restarted the print and it printed just fine. When I looked at the Gcode I noticed that the bed and extruder heat commands were placed AFTER the start GCode from the Printer section. Can you tell me how to correct this or what I am doing wrong?

### Project file & How to reproduce

[608v2_0.3mf.zip](https://github.com/supermerill/SuperSlicer/files/7974633/608v2_0.3mf.zip)

### Version

2.3.57

### Operating system

MAC OS Monterey 12.2

### Printer model

Kingroon KP3S using stock board with Klipper Firmware

supermerill, 2022-02-01:
bad config.
Please use one of the pre-defined one, or the default one.
If you're using one from somewhere on the internet, it may not work with your printer.

jfkerekes, 2022-02-01:
So I set the pri ter up as a custom printer because my printer ter was not
in the predefined ones. Your saying a custom pri ter is not going to work?

neophyl, 2022-02-01:
Not unless you add in EVERYTHING that your custom printer is expecting to the start gcode/end gcode etc.  

Thats the thing about customs, you the user has to add and supply all that information into the profiles.

supermerill, 2022-02-01:
you have written
```
G28 ; home all axes
G1 Z5 F3000 ; lift
G1 X5 Y10 F1500 ; move to prime
G1 Z0.2 F3000 ; get ready to prime
G92 E0 ; reset extrusion distance
G1 Y80 E15 F600 ; prime nozzle
G1 Y100 F5000 ; quick wipe
```
in your start gcode.
That won't work as you don't heat before extruding.

jfkerekes, 2022-02-01:
Got it! Thanks

kadorito, 2022-05-23:
Hello, I want to jump again on this one : my printer standard profile includes some print in the start gcode, and I believe it's a common thing, in order to clean the nozzle at least. The issue is that the hotend is still cold because there is no M109/M190 before the start gcode, hence it fails and messes up everything.

I checked with cura standard code as a comparison, and here is the part that cura inserts before the start gcode (which includes M109/M190).

I guess it would be wise to do the same.

;FLAVOR:Marlin
;TIME:239
;Filament used: 0.133469m
;Layer height: 0.2
;MINX:111.2
;MINY:107.1
;MINZ:0.3
;MAXX:118.8
;MAXY:122.9
;MAXZ:6.1
;Generated with Cura_SteamEngine 5.0.0
M140 S60
M105
M190 S60
M104 S200
M105
M109 S200
M82 ;absolute extrusion mode

neophyl, 2022-05-23:
You say your 'printer standard profile' , well what profile ?  you give no details.  Which profile have you installed and where did it come from ?
As mentioned above you have to add the required gcode for your printer into the profile you are using.  Now the profiles installed via the wizard have these all this set up for you (generally) which is why which one you installed is required information.  
But only if you installed the correct one for your printer.  
Without further information its going to be hard to help you with specifics.

In general though many of the profiles have something like the following to handle the initial heating
M104 S[first_layer_temperature] ; set extruder temp
M140 S[first_layer_bed_temperature] ; set bed temp
M190 S[first_layer_bed_temperature] ; wait for bed temp
M109 S[first_layer_temperature] ; wait for extruder temp

The fields inside the brackets are the placeholders for the relevant temps from the filament settings.  So you dont set bed to S60 in the startup (unless you want to) as with the placeholder it will pull the bed temp from whatever is set in your filament profile.
Of course there are many variation on these depending on if you mesh level, want no ooze startup etc etc but those are the basics you would need to add to your custom start gcode.  I suggest you install a few printers from the wizard temporarily and get a feel for what is used and then ask if you have specific questions.

Also saving a project from within Super Slicer (to generate a .3mf file) and then zipping it up and attaching it would be the most helpful way to get further advice.  Must be zipped or the file attachment is not allowed.  A project will contain copies of the profiles you are using and will allow anyone to more easily check and see where/what could be causing your issues and how to fix them.

kadorito, 2022-05-23:
Thanks for the answer

its' a wanna profile, provided by the vendor.

neophyl, 2022-05-23:
As I mentioned, if you post a zipped up project we can take a look.  Without the profile you are using its hard to say.

Super Slicer will add missed lines into the start gcode automatically but NOT always.  That behaviour can be disabled as many users do not want it doing that.  Also where it adds them sometimes leaves much to be desired.  Theres an option in the printer profile Custom gcode area for Only custom Start G-code.  If your wanhao profile has that ticked then it will only include whats in there.

neophyl, 2022-05-23:
I didn't say a zipped gcode file.  I said a saved PROJECT file.  When you open SuSi then just add anything to the plater like a cube for example, select your printer/filament/print profiles as usual as if you were going to slice.  Then use File>Save Project as.  
This will save a .3mf file.   3mf is the format that Prusa Slicer/Super Slicer use for their projects.  
As well as the object(s) on the plater it will also save a copy of the selected profiles you are using as well as any modifiers etc that may be in place.  Anyone opening the project will then be slicing with exactly the same settings you have.  Its very handy for allowing people to see exactly what is happening with all sorts of issues.  We can check slice previews, generate the gcode if needed and in this case easily see what's in all the different gcode areas as well as everything else.

Yes the heating lines get added after your start gcode if you don't manually add them (unless that option is ticked).  If you want them at a specific position then you need to modify your start gcode block to put them where you want them.  As soon as you manually add them in there then they wont be automatically added afterwards.  I did mention that their placement can be less than ideal.

If your start gcode block was completely empty then there wouldn't be an issue as it would heat up and start printing the gcode that has been sliced from your objects, but as its got SOME custom code in there anyway then its assumed that the custom code will take care of everything it needs to to run that code.  Whoever added the clean nozzle code should have also added the warm up too.