supermerill/SuperSlicer #2295: Right Click Improper M600

BuddyBing, 2022-02-01:
### Version
SuperSlicer
Version:   2.3.57.9
Build:     SuperSlicer_2.3.57.9

### Operating system type + version
Operating System:    Macintosh
System Architecture: 64 bit
System Version:      macOS Big Sur Version 11.6 (Build 20G165)
Total RAM size [MB]: 34,360MB
OpenGL installation
GL version:   2.1 ATI-4.6.20
Vendor:       ATI Technologies Inc.
Renderer:     AMD Radeon Pro 5500M OpenGL Engine
GLSL version: 1.20
Installed extensions:

### Behavior
It appears that when you right click to add a color change, that it does not properly handle the retraction settings associated with that change.  I have a simple M600 filament change macro in Klipper and am receiving an extrusion max error.

When I add a color change using the "+" it properly adds the retraction settings after the M600 (see below gcode output):
;AFTER_LAYER_CHANGE
;17.2
;COLOR_CHANGE,T0
M600
; INIT printing object Voron_Design_Cube_v7.stl id:0 copy 0
G1 E-0.7 F1800
G1 Z17.4 F9000

If I right click and then hit "Add color change" it does not properly insert the retraction settings (see gcode below):
AFTER_LAYER_CHANGE
;6.6
;COLOR_CHANGE,T0
M600
; INIT printing object Voron_Design_Cube_v7.stl id:0 copy 0
; printing object Voron_Design_Cube_v7.stl id:0 copy 0
G1 X119.07 Y107.192

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
[M600 Right Click Issue.3mf.zip](https://github.com/supermerill/SuperSlicer/files/7979108/M600.Right.Click.Issue.3mf.zip)

supermerill, 2022-02-01:
It just insert the gcode from `Colour Change G-code` setting

BuddyBing, 2022-02-01:
> It just insert the gcode from `Colour Change G-code` setting

My 'Colour Change G-code' is just an "M600".  It looks like it is not properly processing a retraction state after a "right click" where you can either add "Add color change (M600)" or "Add Custom G-Code".  This causes a massive extruder move post color change.

supermerill, 2022-02-01:
first color change:
[image]
the filament should be fully insterted in the nozzle after the m600, so no unretract needed. The gcode viewer doesn't understand that, though. Note that I advise for a high skirt for a good prime of the nozzle after a M600.

second:
[image]
Here it tries to unretract after the color change, but as it's not in relative mode, it fails. Something to fix. I think as prusa only use relative mode, they may have that unstested.

BuddyBing, 2022-02-01:
So should I just run in "relative" mode for my E for the time being?

supermerill, 2022-02-01:
yes, it should work.

closed 2022-04-01 by supermerill