supermerill/SuperSlicer #2296: MAC M1- UI blank for  side panel, sliced preview, and Gcode after slicing model

i1986o, 2022-02-01:
**Bug description**
MAC M1- UI blank for  side panel, sliced preview, and Gcode after slicing model,
When slicing a model, after slicing is complete, the side panel info goes blank, as does sliced preview and gcode preview windows, to fix this, you must expand and collapse the side bar, but this does not correct the gcode preview window, it is left blank. 

**Expected behavior**
The windows and side panel should show available content vs being blank after slicing

**To Reproduce**
Steps to reproduce the behavior:
1. open application on mac arm 
2. slice a model
3. side panel content goes blank, gcode preview and slicer preview goes blank
4. to fix panel view, collapse and re open side panel and information will appear.

**Screenshots**
[https://imgur.com/a/R5LyqHz](https://imgur.com/a/R5LyqHz)

**Desktop (please complete the following information):**
 - Mac - Monterey 12.1 (21C52)
 - Apple M1 Pro
 - 2021 macbook pro 16"
 - SuperSlicer 2.3.57

supermerill, 2022-02-01:
Don't use the panelled layout. It's buggy on macos.

allyouaskfor, 2022-02-13:
Similar situation on 2015 MacBook Air, macOS Big Sur 11.2.3. Collapsing and expanding the side panel also fixes the Gcode tab though.

Change the GUI to use "Legacy layout" as it is very similar. To change between the 3D platter view and a sliced view, click the buttons in the bottom left corner.

supermerill, 2022-02-13:
Note that Prusa has created a custom-made Tab widget to be able to use a dark mode on windows. If I switch to it also for the other os, the layout should be fixed. The disadvantage is that everybody will get windows-style tab buttons...