supermerill/SuperSlicer #2301: New version check seems to have issue with last position

RealDeuce, 2022-02-01:
### What happened?

Built and ran 2.3.57.10 when 2.3.57.9 is the latest release... SuSl says there's a new version available.

### Project file & How to reproduce

Build current Nigthly, run.

### Version

2.3.57.10 Nigthly

### Operating system

FreeBSD 12.3-RELEASE

### Printer model

Voron 2.4 with Stealthburner

RealDeuce, 2022-02-01:
[image]

RealDeuce, 2022-02-01:
```
> grep -r version_online .
./SuperSlicer.ini:version_online = 2.3.57.9
```

RealDeuce, 2022-02-01:
Oh, this is on purpose maybe?
```C
                //if (*Semver::parse(SLIC3R_VERSION_FULL) < *Semver::parse(into_u8(evt.GetString()))) {
                    this->plater_->get_notification_manager()->push_notification(NotificationType::NewAppAvailable);
                //}
```

supermerill, 2022-02-02:
problem is "10" < "9"
and I had no time to tinker with it to parse the string correctly into a valid version number.
The real test is at PresetUpdater:l291 anyway

closed 2022-04-01 by supermerill