supermerill/SuperSlicer #2303: First top layer problem, it is sliced half only

nopbit, 2022-02-02:
### What happened?

Hello all,

I had a weird bug, when the top layer sliced, it is sliced in half. Anybody faced smilar problem?

[image]

Thanks for your answers

Mustafa

### Project file & How to reproduce

This problem appears on layer 5 only.
[small-super-slicer.zip](https://github.com/supermerill/SuperSlicer/files/7985585/small-super-slicer.zip)

### Version

2.3.57

### Operating system

Windows 10

### Printer model

Custom Printer

nopbit, 2022-02-02:
It slices correctly if i choose "ensure vertical thickness".

supermerill, 2022-02-02:
I had a similar thing on another issue.
It's fixed in the latest nightly

closed 2022-04-01 by supermerill