supermerill/SuperSlicer #2304: Getting Blobs => max_gcode_per_second ignored?

alexvonsiebenthal, 2022-02-02:
### What happened?

Getting random blobs on prints because the printer stalls for a fraction of a second before continuing. Figured it might be due to queue overflow, hence adjusted max_gcode_per_second. However, the file size stays exactly the same even though I reduced from 1500 to 500 G1 commands / second
[image]
.

### Project file & How to reproduce

[Shape-Cylinder.3mf.zip](https://github.com/supermerill/SuperSlicer/files/7985604/Shape-Cylinder.3mf.zip)

The syncopes / blobs happen on the innermost cylinder

### Version

2.3.57.9

### Operating system

macOS 12.1

### Printer model

voron 2.4 with afterburner

alexvonsiebenthal, 2022-02-02:
Well, I missed part of the information, so here we are:
Apparently the blobs still happen, even when going with 500 G1/second.

supermerill, 2022-02-02:
> However, the file size stays exactly the same even though I reduced from 1500 to 500 G1 commands / second

that is because the input model isn't defined enough to exceed the 500 G1/sec (just a little at some points)

> Apparently the blobs still happen, even when going with 500 G1/second.

reduce it more. 250 G1/sec doesn't seems to reate any artifacts.

alexvonsiebenthal, 2022-02-02:
Thank you. Some more insights I gained meanwhile.
Setup is Voron 2.4. I used Octoprint and observed these random syncopes only with gcode from Superslicer, Prusaslicer and Cura worked just fine. Since the RPi is only a 1G 3B+ model, I switched to mainsail and gave the Superslicer gcode another try and it worked.

Not saying it only has to do with either Superslicer or Octoprint. But it seems that Superslicer generates more demanding code than the other slicers that may run into bottle necks when using Octoprint (which is very resource demanding) on a constrained Pi.

alexvonsiebenthal, 2022-02-02:
> PrusaSlicer has an hardcoded minimum resolution of 0.0125 which is not mandatory for superslicer (as I give the user the choice, even if it's now less foolproof) Your profile deactivate the resolution, and so you have a much more demanding gcode, as intended.

Ah, I missed that one. Thank you! I found some more unexpected things, I'll report them separately. Feel free to close them if they are irrelevant.

closed 2022-02-02 by alexvonsiebenthal